motioneye-project/motioneyeos #1966: Buildroot error help request

PieBru, 2019-06-22:
Hi, I'm trying to enter more deeply in the project, so I'm tinkering with buildroot ME customization.
I'm seasoned in C, but I never used buildroot before.
For a jump start, I'm trying to build raspberrypi3 default .config, where I only added the paho-mqtt-c client.

I'm on Arch Linux. Buildroot prerequisites seems all satisfied, but I get this error using 20190619 released sources:
```
gdbusauth.c: In function ‘_g_dbus_auth_run_server’:
gdbusauth.c:1302:11: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
 1302 |           debug_print ("SERVER: WaitingForBegin, read '%s'", line);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [Makefile:3633: libgio_2_0_la-gdbusauth.lo] Error 1
```

The same error is reported here: https://git.busybox.net/buildroot/commit/?id=4102db0f7a75ecdef3073dd65a5d2cfd512d984c
It seems that "Newer versions of libglib2 prefix glib-genmarshal, gobject-query, and glib-mkenums with ${bindir}. Unfortunately, this will resolve to the host systems /bin/ directory, which will cause compilation issues if the host does not have these programs. By removing the ${bindir}/ prefix, these programs are resolved in PATH instead."

A) I have glib2 installed on my host system. Did you encounter similar issues before?
B) I think it may be a good idea to build using a dedicated VM, but what OS do you suggest?
Thanks,
Piero

PieBru, 2019-06-23:
I wend a bit further, but didn't get the target.
This time I didn't use Arch, but a most common Kali VM, just updated and snapshot.

Steps to reproduce:
```
# sha256sum 20190619.tar.gz
567740246cf3622dd980528519fdd7fd44e8080892b5457d81095fc6fd2e3d53  20190619.tar.gz
...
# cd motioneyeos-20190619/
# export FORCE_UNSAFE_CONFIGURE=1                (Kali uses root user)
# ./build.sh raspberrypi clean-target
# (./build.sh raspberrypi 2>&1) | tee 190623-1758_build.log
# less 190623-1758_build.log
```

Here is the output, any hint on how to overcome this?
```
...
make[2]: Leaving directory '/root/motioneyeos-20190619/output/raspberrypi/build/libmicrohttpd-0.9.60/po'
make[1]: Leaving directory '/root/motioneyeos-20190619/output/raspberrypi/build/libmicrohttpd-0.9.60'
ESC[7m>>> libwebcam 0.2.5 DownloadingESC[27m
--2019-06-23 12:01:25--  http://freefr.dl.sourceforge.net/project/libwebcam/source/libwebcam-src-0.2.5.tar.gz
Resolving freefr.dl.sourceforge.net (freefr.dl.sourceforge.net)... 88.191.250.136, 2a01:e0d:1:8:58bf:fa88:0:1
Connecting to freefr.dl.sourceforge.net (freefr.dl.sourceforge.net)|88.191.250.136|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2019-06-23 12:01:26--  (try: 2)  http://freefr.dl.sourceforge.net/project/libwebcam/source/libwebcam-src-0.2.5.tar.gz
Connecting to freefr.dl.sourceforge.net (freefr.dl.sourceforge.net)|88.191.250.136|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

--2019-06-23 12:01:29--  (try: 3)  http://freefr.dl.sourceforge.net/project/libwebcam/source/libwebcam-src-0.2.5.tar.gz
Connecting to freefr.dl.sourceforge.net (freefr.dl.sourceforge.net)|88.191.250.136|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Giving up.

--2019-06-23 12:01:29--  http://sources.buildroot.net/libwebcam/libwebcam-src-0.2.5.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.25.211.19, 104.25.210.19, 2606:4700:20::6819:d213, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.25.211.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-06-23 12:01:29 ERROR 404: Not Found.

--2019-06-23 12:01:29--  http://sources.buildroot.net/libwebcam-src-0.2.5.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 104.25.210.19, 104.25.211.19, 2606:4700:20::6819:d313, ...
Connecting to sources.buildroot.net (sources.buildroot.net)|104.25.210.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-06-23 12:01:30 ERROR 404: Not Found.

make: *** [package/pkg-generic.mk:146: /root/motioneyeos-20190619/output/raspberrypi/build/libwebcam-0.2.5/.stamp_downloaded] Error 1
```

Thanks,
Piero

jasaw, 2019-06-24:
@PieBru You don't need to install those mentioned programs on your host machine. You edit `motioneyeos/package/paho-mqtt-c/paho-mqtt-c.mk` to build the host programs as a dependency. If you want to apply the patches that you found (from your link above), put the patch files in `motioneyeos/package/paho-mqtt-c` directory. To force it to rebuild just the mqtt package, try this:
```
rm -rf output/raspberrypi3/build/paho-mqtt-c
./build.sh raspberrypi3 paho-mqtt-c
```

Your 2nd issue with the 404 error means your VM is unable to download a package, libwebcam in this case. Make sure your VM has access to the internet when building.

jasaw, 2019-07-05:
@PieBru Have you resolved this issue?

PieBru, 2019-07-05:
BuildRoot solved, MQTT not yet.
The solution I found more simple and stable is using the Vagrant VM via ssh.
Please note I'm on Arch Linux with VirtualBox.
Here are my raw notes, if you want to add it to the wiki after translating it in English ;)

1. Install Vagrant and its disk addon:
```
sudo pacman -S vagrant
vagrant plugin install vagrant-disksize
vagrant up
```
2. Download the MotionEyeOS source tarball, i.e.:
```wget https://github.com/ccrisan/motioneyeos/archive/YYYYMMDD.tar.gz```
3. Uncompress it: ```tar xzvf YYYYMMDD.tar.gz```
4. Go to the directory where the Vagrantfile is: ```cd motioneyeos-YYYYMMDD/support/mish```
5. Edit the Vagrant file to adjust RAM and disk to your needs: ```nano Vagrantfile```

```
VM_MEMORY=2048
...
vagrant.configure('2') do |config|
 config.vm.box = 'ubuntu/bionic64'
 config.disksize.size = '30GB'
end
```
7. Change also the two lines at the end, from:
```
wget -q -c http://buildroot.org/downloads/buildroot-#{RELEASE}.tar.gz
tar axf buildroot-#{RELEASE}.tar.gz
```
to:
```
wget -q -c https://github.com/ccrisan/motioneyeos/archive/YYYYMMDD.tar.gz
tar axf YYYYMMDD.tar.gz
```
8. Start the VM: ```vagrant up```
9. Wait some minutes while it builds the VM. The first time, Vagrant creates a VM named 'Buildroot YYYY.NN' in your (VirtualBox) VMs directory.
10. Access the VM shell: ```vagrant ssh```
11. You should not need the logged-in user credentials here. Just in case... user=vagrant, password=vagrant
12. Check file system mapping and usage: ```df -h```
13. If you did not find the MotiuonEyeOS tarball here, get it and uncompress: ```wget https://github.com/ccrisan/motioneyeos/archive/YYYYMMDD.tar.gz && tar xzvf YYYYMMDD.tar.gz```
14. Go inside the tarball directory: ```cd motioneyeos-YYYYMMDD/```
15. Check which boards are supported in the downloaded version: ```ls -l ./configs/```
16. [OPTIONAL] Fine tune your build: ```make menuconfig``` then ```Load ./configs/[board]```, save it when done.
17. Build your MotionEyeOS, the first time this step may take several hours: ```time ./build.sh [board]```
Cross fingers. If everything went well, the last line in the build log should be _build successful_
18. Build your MotionEyeOS image, this should be quick: ```./build.sh [board] mkimage```
19. Optionally compress it: ```xz /home/vagrant/motioneyeos-YYYYMMDD/output/[board]/images/motioneyeos-[board].img```
20. Copy/move the just created image to the host PC, i.e. you may use scp or directly access the host computer through the /vagrant directory:
```mv /home/vagrant/motioneyeos-YYYYMMDD/output/[board]/images/motioneyeos-[board].img.xz /vagrant/```
On my Arch Linux notebook the VM /vagrant directory corresponds to host computers folder. To find out yours a way is to search the ```Vagrantfile```:
```'/mnt/sdb2/user1/VirtualBox Vms/motioneyeos-YYYYMMDD/support/misc/’```
21. So, in my notebook file-system, I found the image here:
```'/mnt/sdb2/user1/VirtualBox VMs/motioneyeos-YYYYMMDD/support/misc/motioneyeos-[board].img.xz'```
22. Using the host computer SD card writer, write the image just built on the camera SD card, following the normal procedure, i.e.:
```sudo ./writeimage.sh -i motioneyeos-[board].img.xz -d /dev/sdd -n 'mySSID:myPASSWORD'```
23. Boot the camera using the just built micro-SD.

At the time of this writing, the official wiki "Building From Source" page is here: https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
That's all.
Piero

PieBru, 2019-07-08:
Forgot to mention you can start/snapshot/access/... directly from the VM manager (i.e. VirtualBox) the text-only VM created by Vagrant, loggin in as ```user=vagrant  password=vagrant```

Then ```cd motioneye-YYYYMMDD``` and customize your config, i.e.:
```make menuconfig```
```Load ./configs/raspberrypi3-defconfig```

Here you can add tools, i.e. the **mosquitto client** , which was the first reason I used buildroot.

Piero