motioneye-project/motioneyeos #1968: Most settings disappeared in 20190619 release?

PieBru, 2019-06-23:
Hi Calin,
a freshly built SD from https://github.com/ccrisan/motioneyeos/releases/download/20190619/motioneyeos-raspberrypi-20190619.img.xz used on a Raspi Zero W with a common 5MP camera, the web UI shows the camera video and only 4 config sections: Preferences, Network, Services, Expert settings.
Network settings doesn't reflect the writeimage.sh -n parameter value.

The release isn't labeled as "dev" or experimental, is this intentional?
Thanks,
Piero

PieBru, 2019-06-23:
Partially solved.
The UI by default doesn't ask for login and disables most "admin" config sections.
However, only the Preferences section should be visible by the "user" login, as it was in the previous release. 
I was fooled by the "Expert settings" section. IMO Network, Services and Expert settings shouldn't be accessible to not-admins cam users/viewers.
Piero

ccrisan, 2019-06-23:
Can I see a screenshot of the issue? Also a browser full refresh (clearing cache) could help.

PieBru, 2019-06-23:
Steps to reproduce, I'm on Arch Linux:
1. $ sha256sum motioneyeos-raspberrypi-20190619.img.xz
a0c7bbc17e6ce6c91af4d3a53cdf82df1e543b91964ff1dd7e478a88626f6d31  motioneyeos-raspberrypi-20190619.img.xz
2. $ sudo ./writeimage.sh -d /dev/sdd -n 'ssid:password' -i motioneyeos-raspberrypi-20190619.img.xz
3. Boot the Raspi Zero W using the just imaged microSD.
4. Wait for the assigned DHCP lease, wait some minutes more to complete the (slow) PI startup.
5. Browse using a "New private window" on the latest Firefox (so there is no cache). In my case the address is http://meye-abcca2ca.local/
6. At the web UI login prompt don't type anything, only hit the <ENTER> key.
7. After the local cam image is displayed, click the up-left ME menu and the attached screenshot appears.
Thanks,
Piero

[image]

2lostkiwis, 2019-06-26:
I also have the same issue on an Raspi 3 Model B+ with 8MP CSI camera.

ccrisan, 2019-06-26:
Indeed, this can be reproduced and is a motionEye bug. In short, it displays settings that were previously marked and advanced to a non-admin user.

kjr88, 2020-01-30:
Hi there, is there a workaround for this? I've just encountered the same issue on a pi zero w, running motionEyeOS 20190911
[image]

ccrisan, 2020-02-02:
@kjr88 that's a known bug triggered by the fact that you're not logged in with admin. It's been fixed in motionEye and will make its way into motionEyeOS these days. Until then, just log in with admin and you'll be fine.

closed 2020-02-02 by ccrisan