motioneye-project/motioneyeos #1969: Raspberry Pi 4

Chiny91, 2019-06-24:
Surprising many by the look of it, the [new R Pi 4](https://www.element14.com/community/docs/DOC-92648/l/eben-upton-talks-about-the-new-raspberry-pi-4-model-b?ICID=rpimain-feature-content) is out.   Alongside the extra cpu grunt, the long-desired "full throughput gigabit ethernet" should be very helpful to motioneyeos with multiple cameras.

All in due course, in the fullness of time, naturally.   My existing 2 R Pi with motioneyeos remain ultra-reliable running for months on end unattended, and a third R Pi motioneye cam with the @jasaw AI mod still beating my doorbell, and rock stable.

Nameback, 2019-06-24:
As someone new to the project I am curious to understand better what the current bottlenecks are in the existing hardware and how the new hardware could be best leveraged to make for a better system. 

Is the most common use case currently multiple Pi-Zero or Pi-X(Fast Network Mode) with one master Pi encoding and saving to disk?

If the new h264 encoder on Pi 4 CPU allows for 1080p 30 FPS encoding does that mean we can have the full speed while maintaining the smart features we otherwise lose in "Fast Network Mode"? 

Either way I'm very excited for the new hardware and the existing h264 Fast Network Mode in the current pre-release. Thanks for the hard work!

jasaw, 2019-06-25:
Previously, the bottleneck was on Ethernet throughput, CPU (maybe), GPU H264 encoder, memory bandwidth, USB2 port (if you store your media on external disk).
Based on the benchmarks I've seen so far, the RPi4 removes all the above bottlenecks, except for the GPU part. The GPU has been upgraded, gained HEVC hardware decoder, slight speed increase, but no huge improvement other things.

MrJBSwe, 2019-06-25:
Built for the kill
https://www.friendlyarm.com/image/catalog/description/M4SSD_en_04.jpg

ericjanvanputten, 2019-06-27:
Do I assume right I cannot take the Pi 3 image and we need to wait for the Pi 4 image? :)
Hoping this can replace 2 Pi 3's (one at 40% and the other 95% load)

weiserhei, 2019-06-27:
> Do I assume right I cannot take the Pi 3 image and we need to wait for the Pi 4 image?

I get no hdmi output on RPI4 using motioneyeos and trying to run motioneye on raspbian buster (even on PI3) also fails for me (pycurl error).

Nameback, 2019-06-27:
> > Do I assume right I cannot take the Pi 3 image and we need to wait for the Pi 4 image?
> 
> I get no hdmi output on RPI4 using motioneyeos and trying to run motioneye on raspbian buster (even on PI3) also fails for me (pycurl error).

The GPU in the Pi4 is completely new and different and so is the rendering engine used in Buster. It's going to take real time and work for anything that ran on prior Pi's or versions of Raspbian. It's not even working on RetroPie yet which has significantly more developer input involved. Let's just be patient and wait for them to make a pre-release build. I'm not sure where they are located geographically but they may even have a hard time getting their hands on the hardware.

rotrials, 2019-06-27:
@Nameback check the wiki :blush:

Can motionEyeOS be ported to [insert-your-board-here]?

I don't buy boards just to port motionEyeOS on them. If you really want me to try to port motionEyeOS to your favorite board, feel free to send me one and I'll do my best to make motionEyeOS run on it. Also please note that I won't return it to you, given that once motionEyeOS is available for that platform, I'll use it to test new builds.

PieBru, 2019-07-01:
I would not hurry up to adopt the new Raspi 4.
Not all that glitters is gold, as [Andreas Spiess] pointed out in his video: https://www.youtube.com/watch?v=0diTHAmVbhc
Piero

diegoweb, 2019-07-09:
Can someone tell me if Raspberry Pi 4 would be able to handle 3 or 4 IP cameras in 1080p? I was planning to buy some Dahua Starlight cameras.

ccrisan, 2019-07-15:
The latest nightly build has rpi4 support.

tkmilbaugh, 2019-07-15:
I have tried out the latest nightly build "motioneyeos-raspberrypi4-dev20190715", and was able to boot and configure on a pi 4 with 2GB RAM. The only issue that I have found so far is that it will panic and reboot if I don't connect to a wired network. I have configured the wireless network properly, IMHO:

- I used the same wpa_supplicant.conf file that I have used with 3 other raspberry pi models (zeroW, 3B+, 3A+)
- I placed the wpa_supplicant.conf file on the card before booting for the first time (in that same location as the start.elf file).

ccrisan, 2019-07-16:
@tkmilbaugh thanks for the feedback. I'll try to reproduce myself these wifi problems and get back to you. Let's keep the issue open for now.

Nameback, 2019-07-17:
My Pi 3+ died so I'm using a Pi 4 with the nightly build as my network recorder device. Using the new HEVC introduces dramatically reduced performance. It's under 1 FPS. I have it configured with 2x Network Cameras(Pi Zero W) and using H264/OMX quality and speed is solid. I wanted to adjust my GPU memory allocation since I'm using OMX but I don't see an option to. Does this still need to be implemented on the Pi 4 build?

jasaw, 2019-07-17:
RPI4 does **not** have a hardware HEVC encoder. It only has HEVC decoder. Doing HEVC encoding in software is expected to be extremely slow.

ciggybumly, 2019-07-17:
> I have tried out the latest nightly build "motioneyeos-raspberrypi4-dev20190715", and was able to boot and configure on a pi 4 with 2GB RAM. The only issue that I have found so far is that it will [...]

New user here.  Im having the same issue.  Ive taken the wpa_supplicant.conf from a fresh rasbian install and was working fine.  Im getting the same failed message when trying to use the wifi, wired is fine so far.  I also find, when I put the card back into windows, the wpa_supplicant.conf has disappeared.

rusticranger, 2019-07-19:
I have exactly the same issue as above after following all advice  - using the Pi4 with 4GB and another with 2GB - both fail on the wpa_supplicant.conf as placed in the above - and disappears when putting the card back in Windows10 - if I boot off the Rasbian disk all is good for both wireless and wired.

ccrisan, 2019-07-26:
Fixed via https://github.com/ccrisan/motioneyeos/commit/907cc4dc0a5f60844572b274ee05d129ad4f20cb.

closed 2019-07-26 by ccrisan

kay-jahnke, 2019-07-29:
Just a quick feedback: I'm using the Raspi 4 (2 GB version) with two Trust Webcams. The Rapsi is in the closed box it gets delivered with, so it goes to over 80°C and then throttles. The streams on both webcams run flawlessly though. **Good work on that!**

jlr4u, 2019-08-04:
I am trying to install motioneye on raspbian buster on a 4 gig pi-4 that boots off of a 500 gig sdd.
I was attempting to use https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Step 2 fails because libmariadbclient18 has been demoted so I used libmariadb3 instead.
Step 5 (sudo pip install motioneye) fails not finding /tmp/pip-install-xix646/pycurl/

Is there anything I can do to get this up and running?

jlr4u, 2019-08-04:
> Just a quick feedback: I'm using the Raspi 4 (2 GB version) with two Trust Webcams. The Rapsi is in the closed box it gets delivered with, so it goes to over 80°C and then throttles. The streams on both webcams run flawlessly though. **Good work on that!**
Try this 👍 https://www.amazon.com/dp/B07V1QRXTT/ref=cm_sw_em_r_mt_dp_U_e.SrDb0YC19QX
Mine was running in the upper 70s and with this open side case and fan,  It is now in upper 40s.

FCPercival, 2019-08-28:
> Can someone tell me if Raspberry Pi 4 would be able to handle 3 or 4 IP cameras in 1080p? I was planning to buy some Dahua Starlight cameras.

I need this answer too, but i neet to install 6 cameras in 1080p.
Someone that can reply

h4d3z, 2019-11-17:
> > Can someone tell me if Raspberry Pi 4 would be able to handle 3 or 4 IP cameras in 1080p? I was planning to buy some Dahua Starlight cameras.
> 
> I need this answer too, but i need to install 6 cameras in 1080p.
> Someone that can reply?

I also would love to know this.
Does it run alot better on the 4 ? 2GB ?
If it can do 4 cams on 720 it is also acceptable for me

RafaelK34, 2020-01-17:
Hi,

I would be also very interested how the Raspberry PI 4 performs with MotionEye and 4 IP cameras.
Any feedback is really appreciated… also with other configurations.

Many thanks in advance…

bradleyproject, 2020-01-17:
Hi, 

I am trying to complete my final year project at university, which is a vehicle surveillance system. I have come across motioneyeOS and it has ticked so many boxes in terms of functionality relating to my project. However, one functionality is that it will be able to run off 4G technology, which will use my sim card to provide the portable internet connection. My question is, am i able to use 4G with motioneye or will i be limited to use a WiFi or Ethernet connection ?? If someone could please shed some light on this it would be greatly appreciated. 

I have watched some tutorials on how to set up the wifi and it is done quite easily using wpa_supplicant.conf file. I wonder now is it possible for me to get an SSID for my 4G as i have set the password on the mobile hotspot already. The reason i ask is because i have already located the ip address which the 4G connection uses meaning if done i will be able to assign my cameras under this ip address. Any information shared will be awesome. Many thanks in advance guys and gals

jasaw, 2020-01-18:
The easiest is to get a portable wifi access point with 4G connection. They are quite readily available. Search for Nighthawk M1 MR1100 LTE Router. I've used this particular one before, it's not bad.
All you need to do is to configure your motioneyeos device to join this wifi network.

bradleyproject, 2020-01-18:
Hi thanks a lot for the quick response. This does look like something I could use however, I’m using a sample wpa-supplicant file which allows me to connect to WiFi, the configuration of the file requires me to include my ssid (network name) and password. Before this I tried using my hotspot to get portable connection as it allowed me connect on raspbian, that led me to try figure out the ssid of the 4g SIM card. I phoned my service provider and they informed me only WiFi networks have SSID not 4G SIM cards. Have you encountered this issue also ?

Many thanks again..

David

This email and any attachments are confidential and intended solely for the use of the addressee and may contain information which is covered by legal, professional or other privilege. If you have received this email in error please notify the system manager at postmaster@ulster.ac.uk and delete this email immediately. Any views or opinions expressed are solely those of the author and do not necessarily represent those of Ulster University.
The University's computer systems may be monitored and communications carried out on them may be recorded to secure the effective operation of the system and for other lawful purposes. Ulster University does not guarantee that this email or any attachments are free from viruses or 100% secure. Unless expressly stated in the body of a separate attachment, the text of email is not intended to form a binding contract. Correspondence to and from the University may be subject to requests for disclosure by 3rd parties under relevant legislation.
The Ulster University was founded by Royal Charter in 1984 and is registered with company number RC000726 and VAT registered number GB672390524.The primary contact address for Ulster University in Northern Ireland is Cromore Road, Coleraine, Co. Londonderry BT52 1SA

bradleyproject, 2020-01-18:
Wait the ssid isn’t necessary as there is an Ethernet connection yes?

jlr4u, 2020-01-18:
I found the if you load the Motioneyeos operating system, you will find
better performance and less issues than loading Motioneye on top of
raspbian, which does work, but for best performance I would stick with
Motioneyeos.

Jerry.

jasaw, 2020-01-20:
@bradleyproject You're right that only Wifi has SSID. 4G works very differently depending on your 4G hardware. With the nighthawk portable wifi access point, you just stick in your SIM card, configure the router via its web interface (you can configure the Wifi SSID and password at this point). Now you have a portable Wifi network with internet connection. You can configure your MotionEyeOS device to join this Wifi network by setting the SSID and password in the `wpa_supplicant.conf` file.

If you have Ethernet connection to your portable router, then there is no need to configure wifi on your MotionEyeOS, i.e. no need `wpa_supplicant.conf` file.

Whether using ready-made off-the-shelf parts meet your final year project requirements or not is another question.

Ngiri67, 2020-11-11:
Hi there. I hope someone can point me in the right direction.
So I flashed the latest motioneyeos-raspberrypi4-20200606.img.xz, connected the camera and an ethernet cable and tried to boot my Pi4 8GB. It starts to boot and then I get "This board requires new software" and it stops booting.
Any ideas?

starbasessd, 2020-11-11:
You have to update the eeprom software
Install the latest RaspberryPiOS on an SDCard
Do
sudo apt update && sudo apt upgrade -y [enter]
when that completes, do
sudo raspi-config
Select (6) Advanced Options
Select (A7) Bootloader Version
Select (E1) Use the Latest Version
Reboot when prompted
After reboot, shut down and install motionEyeOS normally.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ngiri67, 2020-11-11:
Thank you starbasessd . I'll give it a go right now.

roger-, 2020-11-11:
> You have to update the eeprom software

I have a remotely connected RPi4, does this mean updating to the latest motioneyeos could brick it until I update the firmware?

starbasessd, 2020-11-11:
Depends. Which motionEyeOS are you running?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-11:
FYI, the Bootloader issue was mostly RPi4 4GB & 8GB  related, with the USB
and Netboot. Pre 20200606, you might have had to update the fixup4.dat and
start4.elf files to get it to boot. 0606 update included the corrected boot
files. However, as with most things, it was a 2 edged sword. If you were
someone to always keep the beta release boot loaders up to date, you had to
replace those 2 files, which broke USB cameras. Thankfully, those issues
are behind up now, and only original (factory issued) Pi4s have the boot
software issue, which is easily fixed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ngiri67, 2020-11-11:
Ok, so I ran through the steps you gave. It took a long time to get all the upgraded files and "diversions" in place. Installed the latest bootloader and then tried a fresh Motioneyeos and still getting that error about the board needing new software. 
Hmm, what to do?

starbasessd, 2020-11-11:
Back in the RaspberryPiOS:
sudo apt update && sudo apt full-upgrade -y [enter]

sudo rpi-eeprom-update [enter]

Does it indicate it needs to update?

If yes, then

sudo rpi-eeprom-update -a

Reboot

Try motionEyeOS again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-11-11:
Also, you can try dev20201026 image to see if it boots normally.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ngiri67, 2020-11-11:
Thank you. The eeprom was up to date and it still didn't work. Then I tried the dev version and it booted immediately.

Now to try it out.