motioneye-project/motioneyeos #1970: Cant uninstall motioneyeOS

melikhammoud, 2019-06-26:
Im working on a project for my graduation and i was thinking about using motioneyeOS but it didnt work for me... Now i dont know how to remove it from the SD card so i can get access to the raspbian terminal and use it normally again.... Do i have to format the SD card and install raspbian again or is there anyway to remove motioneyeos only? Im new to all of this, its my first time using raspberry pi....

jasaw, 2019-06-26:
Download Raspbian image and flash it on the SD card. There's no need to format the SD card.

closed 2019-06-26 by jasaw

melikhammoud, 2019-06-26:
but ill have to update it again and download all the apps that i had previously installed??

melikhammoud, 2019-06-26:
> Download Raspbian image and flash it on the SD card. There's no need to format the SD card.

but ill have to update it again and download all the apps that i had previously installed??