motioneye-project/motioneyeos #1971: Motioeneyos + battery + pir => fast boot ?

MrJBSwe, 2019-06-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

MrJBSwe, 2019-06-28:
I would like Motioneye to work standalone on battery and record on PIR signal ( like a trail cam )
( similar to #961 )

This is the fastest solution for pi I have seen so far. 
http://himeshp.blogspot.com/2018/08/fast-boot-with-raspberry-pi.html

Any thoughts ? Tips ?

My thoughts so far :
[Fast CPU, Disk ](https://www.friendlyarm.com/image/catalog/description/M4SSD_en_04.jpg)
Network could be started later

jasaw, 2019-07-05:
I question the viability of running Raspberry Pi off battery. Raspberry Pis are way too power hungry to be practical.

I would look into this project. https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/
It would be really easy to add a PIR. Lower cost too.

MrJBSwe, 2019-07-06:
>esp32
You have a point there,
Sipeed M1 is an other lowenergy option
https://youtu.be/BGctumZuhao

But, as trail-cam I still think a raspberry clone cold be interesting..which means fast-boot has to be solved
( or "too power hungry to be practical" is true...)

ncouture, 2022-01-09:
> I question the viability of running Raspberry Pi off battery. Raspberry Pis are way too power hungry to be practical.
> 
> I would look into this project. https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/ It would be really easy to add a PIR. Lower cost too.

There is no saying how long a motion-fastboot-activated Pi or Pi zero could live on without knowing how much activity it would see and for this reason I think the viability of this feature cannot be questioned.

There's also no saying how much power it has in reserve without knowing how much battery it has and it can potentially have a very large battery.

It may not fit all use cases but will likely fit many.