motioneye-project/motioneyeos #1974: Movie recordings quality is garbage

haha150, 2019-06-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: latest

### Board Model

I am using the following board/model: Raspberry PI 3B

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

External Hard Disk

So i have streaming enabled and movie recording on motion detection and they are saved to a samba mounted drive. Whenever it records a movie and saves it, the quality of the video is complete garbage compared to the streaming video quality, why is that?

i record in mkv/omx
movie quality 100%
stream quality 75%
fps 30

jasaw, 2019-07-01:
Please see https://github.com/ccrisan/motioneyeos/issues/1900#issuecomment-504278190.

closed 2019-07-01 by jasaw

ilovelinux, 2019-07-12:
Try to lower movie quality a bit

rotrials, 2019-07-12:
@ilovelinux for sure it work correctly if you decrease the **resolution** (this is what I have done), but there is a bug in motion... see the issue mentioned by jasaw

ilovelinux, 2019-07-12:
My fault, I didn't read he use omx. I thought it was related to issue 459