motioneye-project/motioneyeos #1975: Enhancement: Reset to factory default settings

PieBru, 2019-06-30:
Hi Calin,
near BACKUP and RESTORE buttons it would be nice a FACTORY RESET, which resets all settings via UI, hopefully with an extra option to keep/modify the wifi/IP/modem configuration ;) without physically access to the PI and remove+reinstall the SD.
This could also be a command or script we can call from ssh, followed by a sync && reboot.
Thanks,
Piero

jasaw, 2019-07-01:
Something like this:
```shell
test -n "${OS_VERSION}" || source /etc/init.d/base
# remount boot with write permission
mount -o remount,rw /boot
# backup wpa_supplicant.conf file
tar -acf /boot/backup-etc-${OS_VERSION}.tar.gz -C /data etc/wpa_supplicant.conf
# remove data partition
echo -e "d\n3\nw" | fdisk /dev/mmcblk0
# reboot and hope for the best
reboot
```

I wrote this without any sort of testing. Please test it on a sacrificial device first.

jasaw, 2019-07-01:
Also see https://github.com/ccrisan/motioneyeos/issues/1756

PieBru, 2019-07-01:
@jasaw very good hint.

I confirm it seems to work, manually executed via ssh on Raspberry PI 3B+ and Zero.
I saved it as a script in /boot/factory-wifi.sh
I'm using it, apparently without problems, but at the second run $OS_VERSION became 'unknown'. This may be an issue when checking for online updates.

As @jasaw pointed out, such a feature reduces the need to disassemble the camera, remove and format the sd.

Some cameras may be difficult to reach physically. One of my cameras is placed just above a bird nest with six azure eggs. It's a Phoenicurus Phoenicurus, and I don't want to disturb those wild birds that choose my patio to build a family.

Thanks, also on behalf of those birds :))
Piero

jasaw, 2019-07-05:
@ccrisan I think it would be beneficial to put a reset to factory default script in the motionEyeOS image and maybe add support to the web interface. What do you think?

jasaw, 2019-07-12:
Future discussions on this topic should go under #1756. Closing issue.

closed 2019-07-12 by jasaw