motioneye-project/motioneyeos #1976: Orange PI Zero H2 512MB with USB camera doesn't boot

PieBru, 2019-06-30:
Hi Calin,
The Orange PI Zero exists also with H5 CPU, for which model is pre-built the MotionEyeOS image?
Sorry for the issue, but the wiki doesn't mention the Orange Pi Zero.
Thanks,
Piero

ccrisan, 2019-07-01:
I have [this](http://www.orangepi.org/orangepizero/) model. which is equipped with Allwinner H2 and that's the board I've been testing the build against. Could I see a photo of your model?

PieBru, 2019-07-02:
Here is my test report. Same PI model here, with minimal differences as you can see on the paper label.

I got the same behavior using two Orange PI Zero H2 512M, one labeled 2017... and the other labeled 2019... v1.4 (see attached pictures).

Used the same micro-SD with your motioneyeos-orangepizero-20190619.img.xz, sha256=57e3ac467c40189717ce95a7a9a0b69f384540edf9aaf6cc9b9ed6572fd099fe.

Both request DHCP lease and MotionEye UI appears only if connected via wired ethernet. WPA* wifi does not work and the PIs remain stuck in a bootloop, hopefully thanks to the connectivity test.

The USB camera is a Logitech C170, it works well with Raspberry PI 3 B+ and A+, but it isn't recognized by both the Orange PIs.

So my tests results are:
1) the wired ethernet works;
2) wifi does not work;
3) USB camera is not recognized,
4) same behavior on earlier (2017) and recent (2019 v1.4) Orange Pi Zero H2 512MB.

Hope this helps to narrow down the issue.
Piero

[image]
[image]

questrov, 2019-08-04:
My test same, USB camera is not recognized on Orange Pi One, both Motioneyeos 20190427 & 20190619.  Same Logitech HD USB cam & powered USB hub working on Raspberry Pi Motioneyeos 20190427.  Additional USB devices (camera & hub, I assume) listed when plug in and lsusb but Motioneyeos add camera, no Local V4L2 camera found before/after reboot...

questrov, 2019-08-05:
Same USB camera and Orange Pi One work OK with Motioneye running on Armbian Buster 
mainline based kernel 4.19.y.  

Nice to have Motioneyeos for the Fast Network Camera on Orange Pi One small form factor.  Happens to fit my subsea enclosure where a Pi 3 will not...

ccrisan, 2019-09-03:
https://github.com/ccrisan/motioneyeos/commit/03b34631dbc8c59cd299bf36f765159e3c7f06c5 adds USB video camera support for Orange Pi Zero and One boards.

closed 2019-09-03 by ccrisan