motioneye-project/motioneyeos #1978: Issue with FTP server acces for uploading

TedSlo, 2019-07-03:
HI,
I have an issue with  FTP server acces for uploading the recorded content to a FTP server on my router (within LAN network) via a motioneye.
I can acces the server via other browser (Filezilla, Firefox...) but when I try the same path, port and password, on the motioneye it won't connect. I get an error massage: 
"Accessing the upload service failed: [Errno -2] Name or service not known! "
I inclouded all the .log files below.
Appreciate any help. 

Thanks!

Ted

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 0.37.1

I am running motionEyeOS version: (motionEyeOS 20171003 ,  0.37.1 ).

### Board Model

I am using the following board/model: (Raspberry PI 3B+).

### Camera

I am using the following type of camera: (Raspberry Pi camera module connected on board).

My camera model is: (Raspberry Pi camera module Rev 1.3).

### Network Connection: 
My motionEyeOS unit is connected to the network via: (WiFi).

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

[motion.log](https://github.com/ccrisan/motioneyeos/files/3355422/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3355423/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3355424/motioneye.log)

stick16, 2019-07-31:
i have the same problem iv also installed ubuntu server with motion eye os on it too and still can't get them to upload to a singular place with a  lot more storage

CubicrootXYZ, 2019-08-20:
Same with SFTP (raspbian with motion eye in docker, raspi 3b+), get error: 
`ERROR: sftp: cURL upload failed on sftp://192.168.178.232:22//files//2019-08-20/18-44-49_2.jpg: Creating the dir/file failed: No such file or directory
   ERROR: failed to upload file "/var/lib/motioneye/Camera1/2019-08-20/18-44-49_2.jpg" with service sftp: (78, 'Creating the dir/file failed: No such file or directory')`

seaniedan, 2020-02-24:
I solved this using the following method:
On the command line, use
sftp user@ftp.site
enter your password, and once connected you should see
sftp>
enter the command to print the working directory:
pwd
take the output of the pwd command and put that absolute path into motioneyeos.

marine1988, 2020-09-18:
Hi there i do have this problem on last version ins dock and can´t do the sft trick any idea?

bohaman1, 2023-01-15:
Hello, May I ask if you were able to get this working? I follow @seaniedan instruction ssh to my motioneyesos then follow the command it susscessfully authenticate with my sftp server and got the output of pwd.

However, when I try to run the test service I still got this error "Accessing the upload service failed: (28, 'Operation timed out after 300015 milliseconds with 0 out of 0 bytes received')!"

Hopefully someone can point out what I'm doing wrong.

Thank you