motioneye-project/motioneyeos #1979: webcontrol interface displays html source instead of webpage

pejotigrek, 2019-07-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry PI Zero

### Camera

I am using the following type of camera: MMAL

My camera model is: generic rpi camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I have my _webcontrol_localhost false_ option set, so I can use port 7999 for alternative motion web interface [I'm controlling few things of my camera via http - from the HomeAssistant].
I updated my motionEyeOS using built-in menu option and suddenly the 7999 web interface started to behave strange.

the interface is no longer rendered as valid webpage [with clickable links, forms to fill settings etc.] and instead, I can see it's source, like this:

    <a href=/><- back</a><br><br>Camera DOORS
    <a href='/1/config'>config</a><br>
    <a href='/1/action'>action</a><br>
    <a href='/1/detection'>detection</a><br>
    <a href='/1/track'>track</a><br>

I'm guessing there's some problem with webserver built into motionEyeOS?

ccrisan, 2019-07-08:
That probably comes from Motion, not motionEye. Use port 80 or whatever you have configured for motionEye.

closed 2019-07-08 by ccrisan

pejotigrek, 2019-07-08:
yeah, I use normal 80 port normally, but 7999 is fot enabling/disabling motion detection "on the fly", what's  used in my smarthome config. it worked fine until I updated motionEyeOS - that's why I'm asking if something changed with the webserver.

JackBradRPI, 2019-07-30:
Same here, on build dev20190728

EDsteve, 2019-11-25:
Something similar happens here too. When i open the web-control i only see this (no links):
```
Motion 4.2.2 Running [1] Camera
1
```
And the is already the page-source. So it's not even html formatted.

I tested with two versions: 20190911 and 20190427