motioneye-project/motioneyeos #1982: Pi 3b+ WIFI connection issues

barneycoles, 2019-07-06:
It seems there are many people having issues connecting to their network with the Pi 3b+ on board WiFi.

I'll be honest I'm no expert.

I have spent an entire day trying everything suggested on the internet in every possible combination.  I have learnt a lot but I'm flat out of ideas.....or am I missing the point?

I have followed all instructions and used different combos of SD cards and Pi 3b+ boards.

I have followed all WiFi related issues on MotioneyeOS GitHub.  I'm not alone and the conclusions seem to be based on sporadic luck and vagueness.

I struggled with this thread.

https://github.com/ccrisan/motioneyeos/issues/537

Raspbian WiFi works great.

MotioneyeOS Ethernet works great.

MotioneyeOS Pi 3b+ WiFi no go.

I'll add that angry ip scanner can't find it but my router admin page is showing a 5 GHz wifi connection to motioneye.  I'm not the first to mention this.

I'm kind of hoping that I am doing something stupid and it's just me being a noob. This is why I haven't put any log files in yet.......it's getting late and I'm beat.

Anyhow, it would be so cool to utilise the two Pi 3b+ boards I have and just have power, board and camera. 

This project is awesome and I really appreciate all the hard work.

barneycoles, 2019-07-07:
LOG update

Fresh build motioneyeos-raspberrypi3-20190427

Text file added as per instruction with default SSiD and PsK of router

country=GB
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="**********"
            psk="***********"
        }

Power up with no ethernet cable and goes into boot loop fail.

Connect ethernet and boots correctly and shows in angry iP scanner and in web browser and router admin page as ethernet device.

Turned on WiFi in browser interface and add SSiD and PsK for DHCP automatic and apply and wait for reboot. All is well again as the ethernet is still connected.

Power down and disconnect ethernet.

Power up and angry iP scanner can't find but my router admin shows meye-*********/ device with an iP address on 5gHz. This address then can't be reached in browser.

Power down and reconnect ethernet to get LOGs

You can see the problem in the penultimate boot in boot log.

I'm not sure how sensitive these logs are? I can provide everything but I think only the boot log is needed. 

BOOT LOG attached

I hope this is enough info and detail?

Many thanks if you have read this.

Again, I hope it's me doing something wrong.

[boot.log](https://github.com/ccrisan/motioneyeos/files/3365588/boot.log)

barneycoles, 2019-08-14:
Sorry I ignored the template.  

This is the latest set up.

 I'm still no further ahead and it's killing me.  

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:    dev20190813

### Board Model    

I am using the following board/model:     Raspberry PI 3B+

### Camera

I am using the following type of camera:        V4L2

My camera model is:   Logitech C270 Webcam

### Network Connection

My motionEyeOS unit is connected to the network via:    WiFi (no go)   and   Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

TP-Link AV500 Powerline  Wired Networking Single Adapter TL-PA411   for Ethernet 

### Log Files

I consider the following log files relevant to this issue:

attached
[boot.log](https://github.com/ccrisan/motioneyeos/files/3502344/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3502345/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3502346/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3502347/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3502348/motioneye.log)

ccrisan, 2019-08-23:
Your `boot.log` indicates no WiFi connection attempt, which is probably due to an invalid or missing configuration file. Make sure a correct `wpa_supplicant.conf` file is present in `/data/etc`.

barneycoles, 2019-08-24:
I'm not sure what you mean by "file is present in /data/etc".

I presume it means this.

[image]

I have followed the Window install and WiFi preconfig instructions creating the wpa_supplicant.conf file on to the SD card after image burn and prior to first boot up.

This is the wpa_supplicant.conf file from Wiki.

country=GB
update_config=1
ctrl_interface=/var/run/wpa_supplicant

    network={
        scan_ssid=1
        ssid="**********"
        psk="***********"
    }

The psk has upper and lower case.

Not sure where I'm going wrong.

barneycoles, 2019-09-24:
Hi. I’m still no further ahead. 

What other reasons could cause the no WiFi attemp? I have done everything as per the wiki for a windows image build. I have tried turning the WiFi on in the web browser user interface which seems to do something but can’t be reached by browser there after. 

I see that some people are having to try all kinds of versions of the wpa_suplicant.conf file. What is the reason for people having to do this? Do the raspberry Pis vary? There is nothing in the wiki about this. How does a noob proceed?

barneycoles, 2020-03-30:
I found the solution finally!

the .conf needs to be upercase  .CONF

closed 2020-03-30 by barneycoles

Brianfit, 2021-04-25:
But it's a read-only file system. How do you change it?