motioneye-project/motioneyeos #1984: Execute Script after detecting Motion

RobinNowak, 2019-07-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

0.40 motionEyeOS Version

4.0 Motion Version

9.9 Raspian (Stretched)

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

Raspberry Pi 3 (B+) Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

Raspberry Pi V2.1 Camera

My camera model is: (insert your camera model, e.g. Logitech C290).

### WiFi
My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

Hello, 
I want to run a command after a Motion is detected.
After reading a lot of differet issues i know that the file should be stored in /data/etc./example.sh 
Unfortunatly i dont have this Directory.

The motioneye.conf file is saved in /etc/motioneye .

Ive installed motioneye os on my Raspberry pi (Raspberry 9.9 Stretched)

I tried putting different paths in to the Run a Command field but the script was never executed even though a Motion was detected and the script is doing its job after executed in the terminal.

(The file was made executable with chmod +x)

Anybody knows where I need to save the script or something else to solve my problem?

kiwi-cam, 2019-07-09:
I think you mean to post this on https://github.com/ccrisan/motioneye/issues.  If you're running Raspbian Stretch, then you are not running motionEyeOS - which is an OS.  That might also explain why you cannot locate the directory as discussed.

It looks like in motioneye (not motioneyeos) it will look in the [configured conf_path](https://github.com/ccrisan/motioneye/wiki/Configuration-File#conf_path) which is `/etc/motioneye` by default.  Try saving there, and if that doesn't work you could raise an issue over on that project - or try the Google Group.

RobinNowak, 2019-07-09:
Thank you! I'll try.

closed 2019-07-09 by RobinNowak