motioneye-project/motioneyeos #1988: Camera image only shown in small window

knivd, 2019-07-10:
I am trying to set up Motioneye OS on rPi3 and it works fine with two noticeable issues:

1. As per the picture here, I can only see the camera output in a small window while the rest of the screen remains unused. Tried all available configuration options but no success so far.

2. The image that I am getting is quite reddish for some reason and I don't know is it possible to apply any colour correction on it.

motionEye Version | 0.40
Motion Version | 4.1.1
OS Version | Raspbian 10
 
[image]

rotrials, 2019-07-10:
2.which kind of camera do you use?

knivd, 2019-07-10:
It is a Sony FCB-EX1020 camera module connected on the CSI port through a PiCapture board