motioneye-project/motioneyeos #1990: After update 20190427 unable to open video device

srekamegawt, 2019-07-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

srekamegawt, 2019-07-11:
[image]

srekamegawt, 2019-07-11:
After update

jasaw, 2019-07-12:
Please provide more information next time you open an issue. Which version were you running before the update?

srekamegawt, 2019-07-12:
Dear Jasaw
English is not my mother's language. 
I live in the Netherlands.
the original program is motioneyeos-raspberrypi3-dev20190223.img.
After the update this problem arises.
I hope that you have sufficient information here.
best regards
Theo

jasaw, 2019-07-15:
Login to your device via SSH, then run `vcgencmd get_camera`, see whether your camera has been detected.