motioneye-project/motioneyeos #1992: Trigger motion detection on/off via console

AndreaCCIE, 2019-07-12:
I've not found a way to trigger the motion detection on/off via console. This could be helpful to integrate the camera with our services (like geofencing).
I can see the setting is in the camera-1.conf -> # @motion_detection off
So I could script the change myself, but I don't know how to restart the service to get the change working.
Is there an easier way?

jasaw, 2019-07-12:
See https://github.com/ccrisan/motioneyeos/issues/842#issuecomment-414375686

Replace PIR with whatever trigger you want to use.

Shaker1978, 2019-12-17:
I am thinking about opening my own issue, but I have the same question as AndreaCCIE. So I'm joining this one here.
#842 doesn't help me, because I cannot find the specific information to solve my problem, which is:

I watched the camera-1.conf file as I made changes in the UI. I turned motion detection ON and OFF. The only value that changed was `# @motion_detection on` (or off). Respectively MotionEye was saving pictures on movement or not.

Now, when I set the value in the config file manually (or a script on a remote machine, which is my primary goal), like from off to on, even after restarting the service with ...:7999/1/action/restart, this value is not applied. Although /action/restart is supposed to read all config files. Executing /detection/start doesn't help either, because after a short while, it's paused again.

So, what what does the WebUI do, when I swith Detection on? Writing to the conf file and restarting the service is not it!

Can somebody help?