motioneye-project/motioneyeos #1995: Intermittent Artefacts in Video Images

bobster316, 2019-07-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 
motionEye Version | 0.40
Motion Version | 4.2.2
OS Version | motionEyeOS 20190619

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Wyze Camera V2

My camera model is: Wyze Camera V2

### Network Connection

My motionEyeOS unit is connected to the network via:  WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
Wyze Camer V2

### Log Files

I consider the following log files relevant to this issue:

[motion.log](https://github.com/ccrisan/motioneyeos/files/3391275/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3391276/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3391277/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3391278/dmesg.log)
 
#### The Issue
When reviewing some of the captured video's fior motioneye i sometimes see artefacts on some of the videos, i will upload 2 examples, the first will be normal and 2nd will show the issue.
I only notice this issue on motioneyeos when i check the equivalent captured file on Wyze Camera I do not see any issues.

Normal Video on Motioneyeos
https://youtu.be/VkRw_8BCQO0

Artefacts Issue on Motioneyeos
https://youtu.be/2v_A1Xp5nvs

jasaw, 2019-07-15:
See https://github.com/ccrisan/motioneyeos/issues/1900#issuecomment-504278190
Please try the test image that I've prepared in the above link and report back whether it fixes the problem for you.

bobster316, 2019-07-16:
Have tried this, now i see the following issue which i was getting intermittently previously.
I can ping the wyse cam from the raspberry pi

[root@MotionEYE ~]# ping 192.168.1.53
PING 192.168.1.53 (192.168.1.53): 56 data bytes
64 bytes from 192.168.1.53: seq=0 ttl=64 time=4.460 ms
64 bytes from 192.168.1.53: seq=1 ttl=64 time=4.729 ms
64 bytes from 192.168.1.53: seq=2 ttl=64 time=4.956 ms
64 bytes from 192.168.1.53: seq=3 ttl=64 time=4.615 ms
64 bytes from 192.168.1.53: seq=4 ttl=64 time=4.518 ms
64 bytes from 192.168.1.53: seq=5 ttl=64 time=5.253 ms
64 bytes from 192.168.1.53: seq=6 ttl=64 time=6.103 ms

motioneyeos does not display any image - "Unable to open video device"

When i try to use filezilla to extract the log files from the wyse cam
i get the following issue

Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
Status:	Waiting to retry...
Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
Status:	Disconnected from server
Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
Status:	Waiting to retry...
Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
Status:	Disconnected from server
Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
Status:	Waiting to retry...
Status:	Connecting to 192.168.1.53:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server

I can still however use the wyze cam app on my iphoine and i can access the camera and all the settings
without any issues..

jasaw, 2019-07-18:
@bobster316 The wyse cam connectivity problem is a different issue. Have you tried connecting Ethernet cable to your Raspberry Pi to see if it helps with connectivity problem?

bobster316, 2019-07-18:
Raspberry pi is connected directly to router port