motioneye-project/motioneyeos #1996: Intermittently losing connection to IP Camera

bobster316, 2019-07-15:
I am running the latest version of motioneyeos with a raspberry pi 3b+ i have been getting multiple issues, (already have another issue open for artefacts) now i have completely lost connectivity to the ip camera (Wyse cam v2) i deleted the camera and tried to reconfigure and get connection refused. 

IP connectivity is fine, i can ping the wyse camera from the raspberry pi.. not sure why this keeps happening..

When checking on the wyse camera app there is no issues, everything is working fine. The only problem is between motioneyeos and the camera.

[messages.log](https://github.com/ccrisan/motioneyeos/files/3392002/messages.log)

jasaw, 2019-07-18:
Have you tried connecting Ethernet cable to your Raspberry Pi to see if it helps with connectivity problem?

bobster316, 2019-07-18:
I am already using eth cable from  router directly to raspberry pi. 
There are no issues with ip connectivity, i can ping all devices from my pc and raspberry  pi.
This issue of "unable to connect to video device" is happening on a regular basis.
[image]

Chiny91, 2019-07-18:
PINGing the cam IP does suggest that the IP address is assigned to something; does the Wyse app show the IP address of the cam ?    What happens if you plug the same URL into something else, a browser, VLC ?   In motioneyeos, have you matched the cam output resolution to the motioneyeos/Video Device/Video Resolution ?   Not to do so, is not necessarily fatal but does confuse my lowest spec R Pi.   The "connection refused" in the log suggests double checking that the cam is config'd to provide rtsp, my cams have other outputs (like http) and rtsp needs activating.   Does another cam work ?

bobster316, 2019-07-18:
Extract from log file from raspberry pi

Jul 18 14:53:10 MotionEYE user.err motion: [1:ml1:Garden] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Garden): Connection refused
Jul 18 14:53:10 MotionEYE user.err motion: [1:ml1:Garden] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jul 18 14:53:10 MotionEYE user.err motion: [1:ml1:Garden] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Garden): Connection refused
Jul 18 14:53:10 MotionEYE user.err motion: [1:ml1:Garden] [ERR] [VID] vid_start: Netcam RTSP failed to open
Jul 18 14:53:10 MotionEYE user.warn motion: [1:ml1:Garden] [WRN] [ALL] motion_init: Could not fetch initial image from camera
Jul 18 14:53:10 MotionEYE user.warn motion: [1:ml1:Garden] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
Jul 18 14:53:20 MotionEYE user.warn motion: [1:ml1:Garden] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
Jul 18 14:53:20 MotionEYE user.err motion: [1:ml1:Garden] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Garden): Connection refused

the wyze cam config does not change, config in the adanced config shows rtsp =  on

I tried to restore config file for the camera in motioneye os, this didnt work

Finally i removed the microsd from  the pi and reloaded the image 
motioneyeos-raspberrypi-20190427.img

bobster316, 2019-07-18:
Still not working

bobster316, 2019-07-18:
using vlc it works straight away

bobster316, 2019-07-18:
completely not working at all from motioneye - keep getting "Unable to open video device" 
vlc is able to playback network stream ok

jasaw, 2019-07-22:
@bobster316 Your IP camera is refusing the connection.
>Jul 15 11:54:12 MotionEYE user.err motion: [1:ml1:Camera1] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera1): Connection refused

What RTSP URL did you configure in your motionEyeOS? Make sure it's correct.

bobster316, 2019-07-24:
The rtsp setup in motioneye is as follows and was workling previously
rtsp://192.168.1.53/live
username: xxxxx
password: xxxxx
Camera RTSP/TCP Camera

bobster316, 2019-07-28:
Still getting the same problem, its really frustrating - unable to open video device

jasaw, 2019-07-29:
This may be an issue with [motion](https://github.com/Motion-Project/motion) software.
Try stopping motionEye and running  motion in the foreground. See https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue

beankylla, 2019-08-27:
is your camera a wyzecam, or Xiaomi Xiafang or other clone? I experienced similar issues with that camera. The issues were resolved with ca firmware update of the camera.

tomg1970, 2020-12-18:
Hello, I have the same problem with INSTAR cameras. The cameras are on latest firmware.

starbasessd, 2020-12-18:
@tomg1970 
Since this issue is over a year old, can you post your setup?
Which Pi, Which version of motionEyeOS?
What URL and type are you using for the Camera Setup?
We would be glad to try to help.

tomg1970, 2020-12-29:
Hello, I installed MotionEye on my PROXMOX server on a Debian VM. After a while this "gray" picture is displayed to me.

[image]

[image]

http://192.168.1.32:84/tmpfs/snap.jpg  **INSTAR IN-7011HD**
[image]

[image]

starbasessd, 2020-12-29:
If you remote into the VM, and run the command:

systemctl restart motioneye

after a few seconds, does the image come back on the WebGUI?

Also this should be posted in : https://github.com/ccrisan/motioneye/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Yes image come back with "systemctl restart motioneye"
And now :)

starbasessd, 2021-01-08:
You can add that command to crontab for a once a day (or however often you
need to) restart.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Difficult - because it can take minutes every few minutes? :(

starbasessd, 2021-01-08:
Shouldn't take more than 10-15 seconds to restart, and you can do it late
at night...
crontab -e [enter]
add at the bottom
* 2 * * * systemctl motioneye restart
 [ctrl]x y [enter]
This will cause motioneye to restart every night at 2AM...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Yes, that's not the problem.

The problem is that sometimes it only takes a few minutes before MotionEye only shows the gray image. And I can't "reboot" every few minutes

starbasessd, 2021-01-08:
Unfortunately, I don't have any spare iron (read: extra PC) to put Proxmox
on) and it doesn't want to work under VirtualBox. Would you try the Docker
install and see if it has similar issues?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Do you have a link to the current Docker, which I can install on PROXMOX?

starbasessd, 2021-01-08:
Here's the link to both BYOD (build your own docker) and Docker image, you
need a VM in Proxmox (I think) to run them.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Yes .... a LXC it´s running on Proxmox in a container.
I have no VM installed

starbasessd, 2021-01-08:
I hate to tell you this, but... Proxmox IS a VM server, and the LXC's are
VMs, just a bit more minimalized than VMWare, VirtualBox, or M$
Hypervisor...
Sorry I forgot the link in the last email...
https://github.com/ccrisan/motioneye/wiki/Install-In-Docker

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomg1970, 2021-01-08:
Yes, just as you describe it is. Thanks for the link. I'll try that

dagababaev, 2021-02-12:
I have same problem with losing connection.

motionEye Version 0.42.1
Motion ver: 4.2.2
OS: Ubuntu 20.04

7 IP (onvif) cameras connected to server and only one camera always lost connection (now permanently not connected), but in same time in VLC stream working properly with same link. All cameras is ZOSI.
"systemctl restart motion" not working for me, motioneye can't connect to camera stream.

[image]

starbasessd, 2021-02-12:
You don't state what kind of hardware/PC you're running it on @dagababaev 
Really, too, this is the motionEyeOS github, you should be posting on the motionEye github
https://github.com/ccrisan/motioneye/issues

closed 2021-02-12 by starbasessd

dagababaev, 2021-02-12:
I'm using VM into ProxmoxVE (Ryzen 5 3500X, 16Gb)
All more interesting… i checked folder where continuously save video and when in UI i saw "unable connect to camera", stream was written. Now i think that problem in UI.

Thank you I repost it to https://github.com/ccrisan/motioneye/issues