motioneye-project/motioneyeos #1998: Save movies as MJPEG?

roger-, 2019-07-17:
My Zero W can only make low FPS H264 clips from an MJPEG source, even with OMX.

Is it possible to save the MJPEG stream directly to a file (e.g. in an AVI container) and bypass the transcoding?

jasaw, 2019-07-18:
What frame rate are you getting with and without OMX?
Without any information about your configuration, I can't give you useful pointers.

Typically with MJPEG, the bottleneck is the wifi network bandwidth. Try changing your source to H264 RTSP and use by-pass recording to save the RTSP stream directly to file.

roger-, 2019-07-18:
Source is  5 fps, output is probably < 1 fps. It's MJPEG, so I can't change it to H264. 

Configuration  is close to stock, with two input sources at about 720p and motion detection on both. Performance is bad even with just one source active.

jasaw, 2019-07-22:
Pi Zero W does not have enough grunt to handle 2 video streams at 720p resolution. Try with 1 stream first, or reduce the resolution.
Your problem could be caused by network bandwidth. Try stopping live stream to your web browser by simply closing your web browser. This should free up some network bandwidth and CPU cycles. Now get it to record some video. When recording is done, check the frame rate in the recorded video.

roger-, 2019-07-22:
Thanks, I've tried all of that and still get a low FPS. I think the transcoding is taking up too much CPU and causing dropped frames. I'm also using Google Photos upload, so that could be part of it.