motioneye-project/motioneyeos #2003: [Google Photos] Accessing the upload service failed: object of type 'NoneType' has no len()!

vedranius, 2019-07-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motioneyeos-raspberrypi4-dev20190715.img - RPi 4 4GB
motioneyeos-raspberrypi3-20190619.img - RPi 3 B+

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

Logitech C290 and few IP cameras

### Network Connection
Ethernet
___________________________________________

Hi, I can't add Google Photos account, or even Google Mail via SMTP.

Mainly, I'd like to get G Photos working.

I get this error when I add a key and click Apply:
"Accessing the upload service failed: object of type 'NoneType' has no len()!"

Tried on both devices on IP and Logitech cameras.

Thanks!

ccrisan, 2019-07-23:
Please provide a full `motioneye.log` file.

karsie1994, 2019-11-09:
Having the same issue log says the following:
   ERROR: gphoto: failed to obtain credentials: Bad Request
   ERROR: gphoto: _get_folder_id_by_name() failed: Bad Request

I tried al different types of folder locations, but nowhere i can find the right information to what structure/location google photos needs.

Sandman224, 2019-11-25:
I have the exact same issue. Would love a solution because Google drive is too small to store my files.

jakepals, 2019-12-16:
When I select Google Photos, it also requires to save it in a directory. I'm not sure how Google Photos is structured. When I test the key, the error "Accessing the upload service failed: object of type 'NoneType' has no len()!" is presented.

ristiisa, 2020-04-15:
> When I select Google Photos, it also requires to save it in a directory. I'm not sure how Google Photos is structured. When I test the key, the error "Accessing the upload service failed: object of type 'NoneType' has no len()!" is presented.

@ccrisan

ccrisan, 2020-04-19:
@ristiisa that error needs to be accompanied by a log file so we can see the full stack trace.