motioneye-project/motioneyeos #2004: Raspberry Pi 3B+ keeps rebooting when WiFi network is off: user.notice network: no network connection available, calling panic action

SiriusFuenmayor, 2019-07-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:  20190427.

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: V4L2.

My camera model is: Microsoft Cam VX-2000.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: USB WiFi adapter

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. messages)
 * (attach e.g. boot.log)
 * (attach e.g. motioneye.log)
 * (attach e.g. dmesg.log)
 
[messages.log](https://github.com/ccrisan/motioneyeos/files/3413878/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3413880/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3413881/motioneye.log)

barneycoles, 2019-07-21:
I have the same issue (#1982) and have spent a day trying to solve.

No response to my issue request yet.

jasaw, 2019-07-22:
@SiriusFuenmayor Your wifi signal may be weak.

Your device managed to connect to your wifi network, but disconnected shortly after.
>Jul 20 04:00:30 android-sirius daemon.info dhclient: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
Jul 20 04:00:32 android-sirius daemon.info dhclient: DHCPREQUEST on wlan0 to 255.255.255.255 port 67
Jul 20 04:00:32 android-sirius daemon.info dhclient: DHCPOFFER from 192.168.1.1
Jul 20 04:00:32 android-sirius daemon.info dhclient: DHCPACK from 192.168.1.1
Jul 20 04:00:32 android-sirius daemon.info dhclient: bound to 192.168.1.3 -- renewal in 721 seconds.
Jul 20 04:00:41 android-sirius user.notice date: current system date/time set to Sat Jul 20 04:00:41 -04 2019 via HTTP
Jul 20 04:11:15 android-sirius user.notice wifi: disconnected
Jul 20 04:11:20 android-sirius user.notice wifi: disconnected
Jul 20 04:11:25 android-sirius user.notice wifi: disconnected
Jul 20 04:11:30 android-sirius user.notice wifi: disconnected
Jul 20 04:11:35 android-sirius user.notice wifi: disconnected for 20 seconds, calling panic action
Jul 20 04:11:35 android-sirius user.notice panic: rebooting (caused by wifi)

Here it managed to connect to your wifi again:
>Jul 20 12:47:04 android-sirius user.notice date: current system date/time set to Sat Jul 20 12:47:04 -04 2019 via HTTP

Check your wifi signal strength by running this on your device:
`cat /proc/net/wireless`

See what wifi quality you get. Level -50 is OK with occasional dropout / lag. Anything below and you'll be struggling to get high resolution video stream over it.

MarkPrinsen48, 2019-08-09:
Just love Motioneyeos but had the same frustrating wpa issues with RPI3B+ and the latest version
Based on this thread, decied to compare image versions with identical ssid/passwd/wpa_supplicant i found out:

Raspbian > wifi works
motionEyeOS_dev20190427 > wifi works!
motionEyeOS dev20190728 > wpa_supplicant failed (brcmfmac)

Maybe worthwhile to check image version differences?
Will be happy to share details if necessary

Version_bootlog results:

[image]

[image]

ccrisan, 2019-08-11:
@MarkPrinsen48 could you please set `OS_DEBUG="true"` in your `/data/etc/os.conf` (using another Linux machine or a wired connection). Then, please attach your `/var/log/wpa_supplicant.log` file.

MarkPrinsen48, 2019-08-13:
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/3498196/wpa_supplicant.log)

MarkPrinsen48, 2019-08-13:
@ccrisan  Hope this helps.
[wpa_supplicant.log](https://github.com/ccrisan/motioneyeos/files/3498206/wpa_supplicant.log)

ccrisan, 2019-08-15:
I believe the relevant part of the log file is this:

    EAPOL: SUPP_PAE entering state DISCONNECTED
    EAPOL: Supplicant port status: Unauthorized
    nl80211: Skip set_supp_port(unauthorized) while not associated
    EAPOL: KEY_RX entering state NO_KEY_RECEIVE

However I have no idea what that means. I would ask you to double check your PSK but since it worked before, that's probably not the problem.