motioneye-project/motioneyeos #2005: Viewing over web doesnt ask for crdentials.

delegating, 2019-07-21:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
20190427

### Board Model 
Raspberry PI 3B+

### Camera
I am using the following type of camera: MMAL
My camera model is: NOIR

### Network Connection
My motionEyeOS unit is connected to the network via: WIFI

### Peripherals
I am using the following peripherals that I consider relevant to this issue:
n/a

### Log Files
n/a

### Issue
I'm viewing my camera over the internet. I've set username and password, but when I go to view it, it doesn't ask for credentials, meaning anyone can view it.
How can I set it to ask for user+pass?

maglub, 2019-07-21:
I would say that this is more a support question, rather than a feature request or a bug report.

delegating, 2019-07-22:
Hi Magnus,

Thanks for getting back to me. Are you able to direct me to the support
forum?

Reagrds,
Calvin

ccrisan, 2019-07-23:
You'll also need to set a password for the surveillance (normal/view-only) user .

maglub, 2019-07-24:
You must have cliecked "remember me" when you logged in, either as admin or "user".

[image]

It is basically impossible to get to the panel view without logging in (even though the default "user" password is empty, you still have to click the ok button).

ccrisan, 2019-07-24:
...and what @maglub said :)

closed 2019-07-24 by ccrisan