motioneye-project/motioneyeos #2007: Thinkerboard (again) :) - no camera image from internal connector

smogger, 2019-07-24:
I am running motionEyeOS version: tried 20190619 / 20190427 / 20190413

### Board Model

Thinkerboard S (OS running from SD card)

### Camera

Camera from USB connection works

Internal camera - unable to load video / model name: Raspberry Pi Camera Rev 1.3 
P5V04A Sunny

Local V4L2 camera shows 4 options, none of them works
Local MMAL Camera shows "no camera"

### Network Connection

Ethernet

### Peripherals

Plain mode, until internal camera not working

lanche11, 2019-07-30:
Same issue with Tinkerboard. Tried the new 0728 OS still the same. I have used Rev1.3,V2.1,UC-261 with Debian V 207 & Motioneye with no problems. The OS gives choices rkisp-statistic for video0 , ME gives rkisp-selfpath for video0. OS works good with any USB 2.0 cam but won't with CSI cam.

ccrisan, 2019-09-02:
If anyone knows what kernel module to enable/load on the Tinkerboard for the CSI camera to work, I'd be happy to enable it by default. I haven't been able to figure it out myself so far.