motioneye-project/motioneyeos #2009: motioneye does not start motion

placidchat, 2019-07-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 0.40rc5

### Board Model

I am using the following board/model: raspi 2B

### Camera

I am using the following type of camera: noir v2

My camera model is: noirv2

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Motion-project version

Self compiled motion v 4.2.2

-------
 
I've altered motioneye to use the new motion version 4.2.2. meyectrl startserver gets to the part where it emits:

`INFO: mjpg client garbage collector started
INFO: server started
WARNING: 403 GET /picture/1/current/?......`

Essentially motion doesn't get started by motioneye. motion itself works fine when doing this: motion -c <motion.conf> , so the problem is motioneye trying to start motion itself.

placidchat, 2019-07-25:
Using motion 4.1.1, and in debug mode, the startup process reaches 
`DEBUG:loading additional config structure for camera, without separators`
`DEBUG: starting motion`

Whereas with 4.2.2 it does not do this at all

maglub, 2019-07-25:
Some missing info:

* How did you compile motion (i.e where, how did you download it, how did you compile it)
* Where did you save the motion binary on your motionos raspberry pi?

To be a stick in your behind, the latest stable release of motionos, 20190427, already has motion 4.2.2. The release you mention in your ticket indicates that you are using the standalone "motioneye", not "motioneyeos". So you might be barking in the wrong repo. The "motioneye" repo/issues are found here: https://github.com/ccrisan/motioneye/issues

On my installation of 20190428 motioneyeos:

```
[root@meye-ce8b29c2 ~]# motion --version
motion: invalid option -- '-'
motion Version 4.2.2, Copyright 2000-2019 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers
```

placidchat, 2019-08-06:
Looks like specifying the specific path solves this.

closed 2019-08-06 by placidchat