motioneye-project/motioneyeos #2010: no network connection available

laurent1981be, 2019-07-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:  motioneyeos-raspberrypi-dev20190715

### Board Model

I am using the following board/model: Rasperry Pi Zero W

### Camera

I am using the following type of camera: MMAL

My camera model is: Joy IT, compatible with CSI port.

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/3432460/boot.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/3432461/dmesg.log.txt)
[hostapd.log.txt](https://github.com/ccrisan/motioneyeos/files/3432462/hostapd.log.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/3432464/messages.txt)

 
Thanks for creating MotionEye, its funny to test and try to run.
But...
I've a Pi Zero W and i can't run the AP mode.
I tried many times and find some errors ont my .conf files. Spaces principaly
Modified now but i already have an error with the "no network connection available"...
Here are my files, i hope you can help me to make my AP run on the Pi Zero W;
Thanks in advance,

laurent1981be, 2019-07-25:
I finally made my AP running by changeing the attirbute in os.conf the deactivate the network and change the wpa with 8 characters, but now, i see a thing ;
If the AP is ON, the detection is OFF,

Is it normal ?

My application of motioneyeOS is to let my pi zero with the cam in my garden and every day (100m from my home, and out of range of my network), go with my smartphone, connect and take back the detection by the AP, but if the detection is OFF, i think it will be impossible...
Thanks a lot

pmatias, 2019-07-25:
I can confirm I'm having the same problem on a Raspberry Pi B+ 2018

pmatias, 2019-07-25:
I just did a clean installation of motionEyeOS (wiped the SD, installed the OS)

Same issue.

pmatias, 2019-07-25:
`wpa_cli` says `CTRL-EVENT-SCAN-FAILED`

`dmesg`:

```
[    8.442681] apple 0003:05AC:024F.0002: input,hidraw1: USB HID v1.11 Device [Matias Matias Notebook Pro Keyboard] on usb-3f980000.usb-1.1.3.4/input1
[    8.556117] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.613739] brcmfmac: F1 signature read @0x18000000=0x15264345
[    8.620969] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.621837] usbcore: registered new interface driver brcmfmac
[    8.860135] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.860274] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    8.860837] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
```

ccrisan, 2019-07-26:
Fixed via https://github.com/ccrisan/motioneyeos/commit/907cc4dc0a5f60844572b274ee05d129ad4f20cb.

laurent1981be, 2019-07-28:
@ccrisan ?
Why do you close my issue ?
It's not the same as @pmatias ?

Thank you,

ccrisan, 2019-07-29:
I wasn't paying attention. My apologies.

GadgetAngel, 2019-07-29:
I ended up going to version 20190427, because motioneyeOS just would fail on my wifi if I used version dev20190728.  I used the same wpa_supplicant.conf file with both versions, (I am using a raspberry pi 3B+ 2017) and when it booted with dev20190728 the wifi fails!

pmatias, 2019-07-29:
same here, I ended up rolling back to `20190427`, I guess I'll now start waiting for stable releases :D

tbird635, 2019-08-07:
Same issue, motioneyeos-raspberrypi3-dev20190728 on a Pi3, I have no wifi connectivity on wifi network with  wpa2 using wpa_supplicant.conf .

My  wpa_supplicant.conf works on my pi zero and on pi3 with 20190427  .

laurent1981be, 2019-08-08:
@ccrisan, some reason ?

laurent1981be, 2020-05-23:
After some month of standing by, a friend of mine call me to build a streamed birdhouse.
That the history...

I found a solution with my PiZ who can't connect to my wifi network, the supplicant file was never good... Don't know why.
The solution I found is the mount a desktop rapsbian on my PiZ (or a other Pi, the process is the same I suppose...), connect to the wifi network. So my PiZ is existing with a good supplicant config... Fantastic !
I toke the file, burn a fresh install of motioneyeOS, put the file in /boot, as explaned in the wiki, put the cables and...
MotioneyeOS can connect to the wifi !!!
I think I can close this issue...
L.

closed 2020-05-23 by laurent1981be