motioneye-project/motioneyeos #2012: Streaming and motion freeze

haha150, 2019-07-26:
I am running motionEyeOS version: motionEyeOS dev20190725

So the issue is that as soon as motion is detected, it does record and save to my external hdd correctly. Then after this is done, both the streaming and motion detection features freeze and i see on the camera clock overlay that the time has frozen, its not ticking anymore after one motion detection. This happened after i updated to this dev version to fix the bitrate issue.

i dont know which log is relevant for you on this.

jasaw, 2019-08-05:
Can you please attach all log files here? motion.log, dmesg, messages, motioneye.log, boot.log

haha150, 2019-08-05:
here you go
[boot.log](https://github.com/ccrisan/motioneyeos/files/3466158/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3466159/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3466160/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3466161/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3466162/motioneye.log)

jasaw, 2019-08-05:
>[1:ml1:Camera-hall] [INF] [ENC] ffmpeg_set_quality: h264_omx codec vbr/crf/bit_rate: 27000000
[0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
[0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!

The bit rate is being set correctly this time, but camera thread stopped responding. How much RAM did you allocate to the GPU? Try lowering the movie quality to see if it happens with lower bit rate. Monitor the GPU memory usage too, using `vcgencmd get_mem malloc ; vcgencmd get_mem reloc` command.

haha150, 2019-08-05:
GPU memory is the default 256

rotrials, 2019-08-05:
I have tested with the same resolution and different GPU mem. size. And no improvements were seen.

jasaw, 2019-08-06:
Try lower bit rate. I wonder whether the GPU locks up when bit rate is too high. @haha150 currently has the bit rate set to 27Mbps. Try lowering it by a factor of 10 just to see whether the GPU still locks up.

haha150, 2019-08-08:
i changed the streaming and movie recording quality both to 75% and it seems to be working fine with that, so 100% quality did just give that error.

jasaw, 2019-08-09:
@haha150 Thank you for the feedback.
I'm going to mark this as a bug for now until I (or someone else) have time to look into it.

jasaw, 2019-08-30:
@haha150 I found that raspivid limits the h264 video bit rate to 25Mbps and you were running at 27Mbps. I think GPU locking up is still a bit harsh. When I get some free time, I'll try to reproduce this problem before reporting to the [Raspberry Pi project](https://github.com/raspberrypi/userland) (unless you want to do it).

For now, make sure your bit rate is below 25Mbps.