motioneye-project/motioneyeos #2017: Couldn't write on etc/wpa_supplicant.conf

esitharth, 2019-07-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20190619).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI Zero).

I am unable to edit the etc/wpa_supplicant.conf. 
The error that I get is : You are trying to save the file on a read-only disk. Please check that you typed the location correctly and try again.

The whole etc folder is closed for editing. How do I add the wifi credentials in ?

ccrisan, 2019-07-27:
See https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf and https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file.

closed 2019-07-27 by ccrisan

esitharth, 2019-07-27:
Hi, 
I did the commands that has been mentioned in the FAQ, still no use. The file still remains readonly.
[image]

ccrisan, 2019-07-28:
Also read the first link. You should edit the `/data/etc/wpa_supplicant.conf`.

esitharth, 2019-07-28:
Hi,
The problem is both the folders are write protected and couldn't get the
wifi_supplicant.conf file in.
writeimage.sh also throws up an error when running it.

Regards,
Sitharth E

esitharth, 2019-07-28:
/dev/mmcblk01 does not exist. is the error that pops up when using
writeimage.sh.

Regards,
Sitharth E