motioneye-project/motioneyeos #2018: Wifi_supplicant for Access point

usbrit, 2019-07-29:
Does anyone have a working wifi_supplicant file A P use on a P0W?
I am struggling to understand what I should be using at the moment.
Thanks

GadgetAngel, 2019-07-29:
if you are trying to just use your wireless card on raspberry pi here is what I used ( but it does not work in the latest version 20190728,  it will work on version 20190427):

NOTE: ensure that your Line feeds in the file are Linux type (I use the editor "notepad++" on windows to do this)
Here is the text for wpa_supplicant.conf:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID NAME"
    psk="your password"
    key_mgmt=WPA-PSK
}

If you think you have the correct information in your wpa_supplicant for Access Point then I suggest your try version 20190427 of motioneyeOS

usbrit, 2019-07-30:
Thanks for this, but tried it and no luck. For some reason motioneyeOS doesn't see the supplicant and looks for a wired network.
Used Notepad++ 
still get the boot loop.

GadgetAngel, 2019-08-04:
Did you try a different version?

jasaw, 2019-08-05:
@usbrit You ignored the issue template and didn't provide any information on what you're trying to do and what problem you're encountering and what your hardware set up is like. Please answer all those questions and perhaps someone might be able to help you.

usbrit, 2019-08-05:
Sorry, how do I do that now the thread is open?
Thanks,
David
Instagram <http://www.instagram.com/usbrit/>

jasaw, 2019-08-06:
Just answer the below questions:
- What are you trying to set up? Getting your Pi to join your wifi network? Or setting up access point on your Pi?
- What problem did you encounter? What error messages did you see?
- Step by step of what you did.

usbrit, 2019-08-06:
Just answer the below questions:
I need to set up access point on your Pi? (as described in the thread description)
This is so I can access MEOS files directly WITHOUT the need for a secondary network

What problem did you encounter? What error messages did you see?
After following the instructions here: https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point (files created with Notepad++ with linux eol setting) I get a panic reboot at the line wpa_supplicant, since no supplicant file causes a search for a wired network that does not exist on a pi zero W.

Step by step of what you did.-As above, followed all the instructions, always reformatted and etched image each time I tried. I can join the pi zero W to a network without problems, so I know my process is good, just need to no how to circumvent the search for a wired network.

Thanks,
David

jasaw, 2019-08-06:
I presume you have created `/boot/hostapd.conf` and `/boot/dnsmasq.conf` by following the instructions. If you don't want the Pi to connect to an existing network, you'll need to set `OS_NETWORKLESS=true` in `/boot/os.conf file`. It's easy to miss this step in the wiki. For more info about `os.conf`, read this: https://github.com/ccrisan/thingos/wiki/os.conf

usbrit, 2019-08-06:
OK, Thanks, didn't see that. So that must be done after first boot? Can I edit the file from command line in MEOS via SSH?
Regards,
David

usbrit, 2019-08-06:
So... how to edit os.conf on a Pi Zero W with no network?? Can't access the boot partition from Windows or Ubuntu. Can't SSH/Putty (no network)
Do I have to:

- Boot on Pi Zero W to create partition.

- Insert into USB adaptor then into Pi that's already running?

- How else to do it, please??

Thanks,
David

jasaw, 2019-08-06:
If you have a Linux machine, you can pull the SD card out and edit the /data/etc/os.conf file directly.
Otherwise, connect your Pi to Ethernet and SSH into it to modify /data/etc/os.conf
Last resort is to reflash the SD card and create /boot/os.conf **before** first boot.

usbrit, 2019-08-06:
Now boots and reports IP as 192.168.27.1/24
But can't see a network or connect to it as a hidden network from PC or android phone.
Can anyone see any issues please?
TIA 
David

Here's my hostpad.conf:

> ssid=s******1
> wpa_passphrase=c******h
> wpa=2
> wpa_key_mgmt=WPA-PSK
> interface=wlan0
> channel=1
> driver=n180211
> hw_mode=g
> ieee80211n=1
> ieee80211ac=1

Here's my dnsmasq.conf:

> interface=wlan0
> dhcp-range=192.168.27.50,192.168.27.150,24h
> dhcp-leasefile=/data/dnsmasq.leases

Here's my os.conf:

> OS_DEBUG="false"
> OS_PRERELEASES="false"
> OS_TTY_LOGIN="tty1"
> OS_ETH="eth0"
> OS_WLAN="wlan0"
> OS_PPP="ppp0"
> OS_NETWORKLESS="true"
> OS_COUNTRY="GB"
> OS_FIRMWARE_METHOD="github"
> OS_FIRMWARE_REPO="ccrisan/thingos"
> OS_FIRMWARE_USERNAME=""
> OS_FIRMWARE_PASSWORD=""

ccrisan, 2019-08-07:
@usbrit how did you end up having `OS_FIRMWARE_REPO="ccrisan/thingos"`? I'm not saying it has anything to do with your issue, I'm just trying to figure out where can that setting be found on motionEyeOS.

usbrit, 2019-08-08:
@ccrisan I needed a clean version of the os.conf file as I could not edit the /data/etc/os.conf (no linux access), so under File Location And Role on this page: https://github.com/ccrisan/thingos/wiki/os.conf#os_networkless
I found the link to os.conf and copied file from there to add it to the /boot before first boot.
https://github.com/ccrisan/thingos/blob/master/board/common/overlay/etc/os.conf

Still no luck accessing as an access point. Is there something I need to do to broadcast the SSID?

Thanks

tachas, 2019-09-20:
I made a motioneyeos-to-go, to use as a dashcam using rbp zeroW. After some failures, I succeded. Here are the steps I followed:

- Flashed a new SD card with the pre-realease image motioneyeos-raspberrypi-dev20190829.img (which I cannot find now, but I guess the stable release would be ok).
- BEFORE the first boot, I places 4 configuration files in the boot partition (practically copy-pasting from the wiki): [os.conf (to set networkless to true)](https://github.com/ccrisan/thingos/wiki/os.conf), [dnsmasq.conf](https://github.com/ccrisan/thingos/wiki/dnsmasq.conf), [static_ip.conf](https://github.com/ccrisan/thingos/wiki/static_ip.conf) and [hostapd.conf](https://github.com/ccrisan/thingos/wiki/hostapd.conf). 
- Needless to say, the static_ip and dnsmasq network must be in the same ip range (for instance static_ip: 172.16.0.1/16 dnsmasq: 172.16.0.100,172.16.0.150 -basic networking theory- )

That way, I was able to join my new AP and access the UI through the browser on 172.16.0.1 and configure the options available, like the timezone, and video settings.
However, since I never connected to the Internet, the date was set in 1970 and so the recording files were not labeled correctly. 
Besides, I wasn't able to ssh to my device, since my rbp was configured as networkess (even though it became an AP). So, to fix that, I had to plug the rbp to a monitor and a keyboard and do the following:   
- Set manually the date using the (suprisingly) date command.
- Remounted the root partition in RW mode (mount -o remount,rw /) in order to
- Comment out the dependency of the network ssh init script (/etc/init.d/S60sshd): the one that tests if it is networkess and exits.

Voilá, my homemade dashcam is working now.

I hope someone finds this info userful