motioneye-project/motioneyeos #2019: (zero) 0KB recordings and pictures

PhinPhin, 2019-07-29:
I have a working installation of MotionEyeOS here. There is just one issue: the video- and picturefiles are 0KB big... I have no clue what goes wrong there. The logfiles also show no sign of an error.

Maybe somebody had a similar issue and can help me? :)

### motionEyeOS Version
motioneyeos-raspberrypi3-20190427.img

### Board Model
Raspberry Pi 3B+

### Camera

v4l2-ctl -D
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : Microsoft® LifeCam HD-3000: Mi
        Bus info      : usb-3f980000.usb-1.3
        Driver version: 4.14.98
        Capabilities  : 0x84200001

As far as I can see it is running in the YUV420 mode? Any way to make sure?

### Network Connection
Gigabit-Ethernet

### Peripherals
Networkshare (SMB1.0) on a Fritzbox 7490, it's accessible from the Pi and I can write files on it

### Log Files

Nothing to see here, there's just one IP-camera missing... nothing obvious....

 * (attach e.g. motioneye.log)

`[2:ml2:Kueche] [ERR] [NET] netcam_connect: connect returned error: Operation now in progress
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to open camera - check your config and that netcamera is online
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[2:ml2:Kueche] [ERR] [VID] vid_start: Netcam failed to open
[2:ml2:Kueche] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[2:ml2:Kueche] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Kueche] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[2:ml2:Kueche] [ERR] [NET] netcam_connect: connect returned error: Operation now in progress
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to open camera - check your config and that netcamera is online
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[2:ml2:Kueche] [ERR] [VID] vid_start: Netcam failed to open
[2:ml2:Kueche] [ERR] [NET] netcam_connect: connect returned error: Operation now in progress
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to open camera - check your config and that netcamera is online
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[2:ml2:Kueche] [ERR] [VID] vid_start: Netcam failed to open
[2:ml2:Kueche] [WRN] [ALL] motion_init: Could not fetch initial image from camera
[2:ml2:Kueche] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Kueche] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
`

 * (attach e.g. boot.log)
```
---- booting motionEyeOS 20190427 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Configuring wired network: dhcp
 * Setting current date using ntp: Mon Jul 29 14:14:35 CEST 2019
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.219/24
 # Default gateway is 192.168.1.2
 # DNS server address is 192.168.1.2
```

PhinPhin, 2019-07-30:
Just to show it still happens: 

[image]

jasaw, 2019-07-30:
You're using a USB camera, but your motion is trying to reach a netcam?
>[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to open camera - check your config and that netcamera is online 
[2:ml2:Kueche] [ERR] [NET] netcam_http_request: Failed to read first camera header - giving up for now
[2:ml2:Kueche] [ERR] [VID] vid_start: Netcam failed to open

Check your motion configuration.

PhinPhin, 2019-07-30:
Yes, the camera was offline because of a changed wifi-password. Now it is working. Both cams work, but not the recording. I can save snapshots from the webinterface manually but the motion-detect only writes 0Byte files on the SMB-share.

**motion.log:**
`[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_open: Could not set the stream
[1:ml1:Wohnzimmer] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for m                                                                                                                        ovie output.
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/media                                                                                                                        /motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-03-57.mp4: Re                                                                                                                        mote I/O error
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_open: Could not set the stream
[2:ml2:Kueche] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie                                                                                                                         output.
[2:ml2:Kueche] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/m                                                                                                                        edia/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-09-39.mp4                                                                                                                        : Remote I/O error
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_open: Could not set the stream
[1:ml1:Wohnzimmer] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for m                                                                                                                        ovie output.
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/m                                                                                                                        edia/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-10-26.mp4                                                                                                                        : Remote I/O error
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_open: Could not set the stream
[1:ml1:Wohnzimmer] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for m                                                                                                                        ovie output.
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/media                                                                                                                        /motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-10-45.mp4: Re                                                                                                                        mote I/O error
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_open: Could not set the stream
[2:ml2:Kueche] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie                                                                                                                         output.
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/media                                                                                                                        /motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-13-06.mp4: Re                                                                                                                        mote I/O error
[2:ml2:Kueche] [ERR] [ENC] ffmpeg_open: Could not set the stream
[2:ml2:Kueche] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for movie                                                                                                                         output.
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/m                                                                                                                        edia/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-13-25.mp4                                                                                                                        : Remote I/O error
[1:ml1:Wohnzimmer] [ERR] [ENC] ffmpeg_open: Could not set the stream
[1:ml1:Wohnzimmer] [ERR] [EVT] event_ffmpeg_newfile: Error opening context for m                                                                                                                        ovie output.
`
**motioneye.log:**

`
2019-07-30 11:45:44: [motioneye]    ERROR: failed to create movie preview for /data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/09-26-02.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/09-26-02.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/09-26-02.mp4.thumb']' returned non-zero exit status 1
2019-07-30 11:45:54: [motioneye]    ERROR: failed to create movie preview for /data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-29/21-43-00.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/motioneye_19 [... cut ...]
2019-07-30 11:45:58: [motioneye]    ERROR: failed to create movie preview for /data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-29/19-32-52.mp4: Command '[u'ffmpeg', u'-i', u'/data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-29/19-32-52.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'4', u'-y', u'/data/media/motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-29/19-32-52.mp4.thumb']' returned non-zero exit status 1`

So I guess there is some weird stuff with the network-share going on?

jasaw, 2019-08-05:
>[2:ml2:Kueche] [ERR] [ENC] ffmpeg_set_outputfile: Error opening file /data/media /motioneye_192_168_1_2_fritzbox_bilder_opacam_webcam/2019-07-30/17-03-57.mp4: Re mote I/O error

Problem accessing remote machine. Try saving the video on the Pi locally first, see if you still get zero file size recordings.

Rob58329, 2021-10-30:
re. 0KB recordings and pictures.
Just in case others find this thread with a similar issue, I had this 0KB recordings issue for a different reason when running motioneye on a Raspberry Pi Zero.  Specifically everything would work fine for a number of months, and then I would start getting days were all recordings were 0KB.  A reboot would normally fix this for a few days, and they the issue would re-occur.  In my case, this was a sign of a failing microSD-card in the Raspberry Pi Zero.  Changing the microSD-card for a new one fixed it (for another approx 6months).