motioneye-project/motioneyeos #2020: Downgrade quality after 12:00 oclock

fefed22, 2019-07-30:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi 3

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: Night vision camera module for Raspberry Pi

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Screenshot

[image]
[image]

### My issue

I have :
- activate continous recording at 1920x1080 (quality 75%)
- H264/OMX and a max length : 900 seconds
- Store file on a NAS on network

after 12:00 oclock, the quality file is downgrade as i have put quality to 5% !!! but not at all...
did you know this issue ?

PS : as you can see screenshot give you resolution for each video but quality are really different !

jasaw, 2019-07-30:
I think it's the same as this bug: https://github.com/ccrisan/motioneyeos/issues/1900
Please try the latest nightly dev and report back. https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev

rotrials, 2019-07-30:
It looks like the same issue. An fast fix is to decrease the resolution at  1280x720. But anyway, please let as know if the with the latest nighty dev is working.I'm also interested...

fefed22, 2019-07-30:
Have you a way to switch the update checker to dev or nightly channel ?

haha150, 2019-07-31:
> 
> 
> It looks like the same issue. An fast fix is to decrease the resolution at 1280x720. But anyway, please let as know if with the latest nighty dev is working.I'm also interested...

Having issues with the nightly dev which i posted in https://github.com/ccrisan/motioneyeos/issues/2012

fefed22, 2019-07-31:
I make a test whith video recording only AND reduce resolution to 1600x1200...
i come back with some news...

PS : i notice that when i put 1600x1200 i see that the cam zoom OUT !!

fefed22, 2019-08-01:
i have the same issue after changing settings but NOT upgrading to last release (dev).

Have you a way to switch the update checker to dev or nightly channel ?
Or i will erase all data on SD card ?

jasaw, 2019-08-05:
@fefed22 Enable "Enable Prereleases" under "Expert Settings" and try checking for upgrade again. This won't wipe all your data on SD card unless something goes wrong horribly and you have to reformat the SD card.

jasaw, 2019-09-09:
@fefed22 Have you tried the nightly build or prerelease? Low bit rate issue should be fixed by now.

jasaw, 2019-10-30:
No update from @fefed22 . I'm going to assume it's recording bitrate issue which has been fixed.

closed 2019-10-30 by jasaw