motioneye-project/motioneyeos #2025: Camera preview keeps going offline. Reboot required.

Mike-Purcell, 2019-08-03:
### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: 2xRPi Zero W (cameras), 1x RPi3B+ (Hub)

### Camera

I am using the following type of camera:2x RPI Noir V2 on the RPI Zero's

### Network Connection

My motionEyeOS unit is connected to the network via: RPi Zeros connected to home wifi and wifi extension. RPi 3B+ connected via ethernet to router via Gigabit Ethernet Switch

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

WD 1TB USB Hard drive connected to RPi 3B+ as storage for files from RPi zeros. The RPi zeros store locally but upload to RPi Hub.

### Log Files

I consider the following log files relevant to this issue:

These files are dared: 2019 08 03 1340 GMT
[boot.log](https://github.com/ccrisan/motioneyeos/files/3463833/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3463834/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3463835/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3463836/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3463837/motioneye.log)

After a period of time when viewing my "RearCam" via either Hub or directly the camera preview seems to go offline however, I still have recording, suggesting that the camera is still operating but the preview goes off.

Mike-Purcell, 2019-08-03:
Currently attempting a re-build of the SD Card. 

PS. MicroSDHC Card in use: SanDisk Edge 16Gb A1 Class 10

Mike-Purcell, 2019-08-04:
Re-build attempted. No improvement. Fresh logs attached. Any help would be greatly appreciated.

[boot.log](https://github.com/ccrisan/motioneyeos/files/3464884/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3464885/dmesg.log)
[messages-2.log](https://github.com/ccrisan/motioneyeos/files/3464886/messages-2.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3464887/motion.log)
[motioneye-2.log](https://github.com/ccrisan/motioneyeos/files/3464888/motioneye-2.log)

Mike-Purcell, 2019-08-04:
Error just prior to camera going offline:

Aug  4 18:16:50 meye-e7677bae daemon.info ntpd[1076]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Aug  4 18:23:18 meye-e7677bae user.err motion: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
Aug  4 18:23:28 meye-e7677bae user.err motion: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
Aug  4 18:23:29 meye-e7677bae user.err motion: [0:motion] [ERR] [ENC] ffmpeg_flush_codec: Error writing draining video frame
Aug  4 18:23:29 meye-e7677bae user.err motion: [0:motion] [ERR] [ENC] ffmpeg_close: Error flushing codec

jasaw, 2019-08-05:
You're getting kernel oops during vchiq_bulk_transfer, i.e. video-core related error, and camera goes through video core. Kernel was dereferencing a NULL pointer at some random address. In another instance, kernel was dereferencing a pointer at address zero. You have a damaged Pi, or your power supply is severely under spec'ed.
> [ 5702.611620] Unable to handle kernel NULL pointer dereference at virtual address 00000264
[ 5702.611636] pgd = d69f4000
[ 5702.611643] [00000264] *pgd=16b6d831, *pte=00000000, *ppte=00000000
[ 5702.611663] Internal error: Oops: 817 [#1] ARM
[ 5702.613591] Modules linked in: ipv6 spidev brcmfmac snd_soc_bcm2835_i2s brcmutil regmap_mmio snd_soc_core snd_compress cfg80211 snd_bcm2835(C) snd_pcm_dmaengine rfkill snd_pcm i2c_bcm2835 snd_timer snd spi_bcm2835 uio_pdrv_genirq uio fixed i2c_dev i2c_bcm2708 bcm2835_v4l2(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media
[ 5702.622563] CPU: 0 PID: 1142 Comm: VCHIQ completio Tainted: G         C      4.14.98 #1
[ 5702.626110] Hardware name: BCM2835
[ 5702.628170] task: d6aed340 task.stack: d6b4c000
[ 5702.630180] PC is at reclaim_clean_pages_from_list+0x184/0x1d8
[ 5702.632251] LR is at shrink_page_list+0x4a0/0x1020
[ 5702.634269] pc : [<c010ee14>]    lr : [<c010e110>]    psr: 00000013
[ 5702.636653] sp : d6b4db28  ip : d6b4da68  fp : d6b4db7c
[ 5702.639057] r10: d6b4daa8  r9 : d6c048d4  r8 : 00000000
[ 5702.641112] r7 : 00000000  r6 : 0001751e  r5 : 00001040  r4 : c09d0e30
[ 5702.643189] r3 : 00000260  r2 : d6b4db30  r1 : d6b4db30  r0 : ffffffff
[ 5702.645671] Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 5702.648531] Control: 00c5387d  Table: 169f4008  DAC: 00000055
[ 5702.651034] Process VCHIQ completio (pid: 1142, stack limit = 0xd6b4c188)
[ 5702.653278] Stack: (0xd6b4db28 to 0xd6b4e000)
[ 5702.655454] db20:                   00000000 00000001 00000260 c09d0e30 00000000 014000c0
[ 5702.659719] db40: 00000000 00000000 00000000 0000000c 00000000 00000002 00000001 00000000
[ 5702.664120] db60: d6b4dba0 00017520 00017520 d6b4c000 d6b4dc14 d6b4db80 c0100ff4 c010ec9c
[ 5702.668470] db80: 00000002 00000006 fffffff0 00017800 00017400 00000004 00000000 00000000
[ 5702.673397] dba0: d6b4dba0 d6b4dba0 c09d0e30 00000000 00000001 00000002 00000000 00000000
[ 5702.678086] dbc0: 00000000 0001751f 014080c0 ffffffff 00000000 00000000 00000000 00000002
[ 5702.683773] dbe0: 00000001 00000000 00000002 0001751e 0000011e c0a595b8 fffffff0 00000002
[ 5702.688807] dc00: 00000002 00000800 d6b4dc64 d6b4dc18 c0156634 c0100ec0 00000001 00000000
[ 5702.694171] dc20: 00000001 00000002 00000000 014080c0 00000001 c0a595c8 00000000 00000001
[ 5702.699563] dc40: d6b4dce8 00000001 00000247 00002000 00000000 d6b4dd94 d6b4dc74 d6b4dc68
[ 5702.705505] dc60: c040e070 c0156540 d6b4dc9c d6b4dc78 c0019c90 c040e038 d6b4dce8 c0520514
[ 5702.711028] dc80: d7167410 00000001 00000000 d6a7b120 d6b4dccc d6b4dca0 c0019d68 c0019c5c
[ 5702.717470] dca0: c0520514 00000001 00000000 014080c0 ffffffff 014080c0 d7167410 ffffffff
[ 5702.723171] dcc0: d6b4dd34 d6b4dcd0 c0019f28 c0019d24 c004c588 c0646058 c004cb68 c0040b7c
[ 5702.728989] dce0: d6a601a0 00000000 00000000 d7167410 00002000 014080c0 00000247 c0520514
[ 5702.734558] dd00: d6aed301 00000000 00000000 00000000 00000000 000afc80 00000020 000010b4
[ 5702.740638] dd20: c001a108 d6aed340 d6b4dd5c d6b4dd38 c001a154 c0019d7c 00000247 00000000
[ 5702.746506] dd40: 00000000 c0520514 000000b0 d7167410 d6b4ddd4 d6b4dd60 c0520514 c001a114
[ 5702.752708] dd60: 00000000 00000200 c0a839bc c051f1b8 00000000 d6b06350 d6b4dd9c d6b4dd88
[ 5702.758595] dd80: b1e4d020 00000001 00000001 000afc80 d6b4ddd4 ffffffff c051f1c4 c002642c
[ 5702.764556] dda0: c0656428 c065306c d5980030 d6b06200 00000000 b1e4d020 00000000 000afc80
[ 5702.770388] ddc0: 00000072 00000006 d6b4de44 d6b4ddd8 c05187ec c0520420 00000001 c0654cb4
[ 5702.776068] dde0: d6b4de04 d6b4ddf0 c0654cb4 00000060 d6b0633c 00000000 d6b06350 d6b06388
[ 5702.782449] de00: c0a67460 d6b062dc d6b4de2c 20000113 20000113 c09c68b0 b30fecc4 c014c406
[ 5702.788134] de20: c09c68b0 b30fecc4 00000001 d6b06200 d7580194 d69fd000 d6b4df0c d6b4de48
[ 5702.793735] de40: c051d4f0 c0518520 b27095e0 00000000 00000001 c00486a4 d5a25e64 d7123ab4
[ 5702.799327] de60: d6b4de84 d69fd824 b30fed5c c014c407 d7580194 00000130 0000012c d6b06200
[ 5702.805363] de80: d6b4dec4 00000007 ffffffff b3100000 b27095e0 00000000 00000000 00000001
[ 5702.810889] dea0: 0000d004 b1e4d020 000afc80 b27095e0 00000000 d6b4dec0 c0081c48 c0080114
[ 5702.817305] dec0: c004cb84 c0040b7c 00000000 d5980000 00000001 d6aed340 d6b4df1c d6b4dee8
[ 5702.822869] dee0: c0046ccc b30fecc4 d730bd88 d58d68c0 00000007 00000007 d6b4c000 00000000
[ 5702.828540] df00: d6b4df7c d6b4df10 c016d9dc c051cfa8 d6b4df74 d6b4df20 c0652bf4 c0046c7c
[ 5702.834081] df20: 00000001 00400040 00083d10 0000000a 793525c0 c017873c d699aa80 b6de4654
[ 5702.840149] df40: b30fecc4 000afc80 c014c406 00000007 d6b4df6c d58d68c1 b30fecc4 d58d68c0
[ 5702.845915] df60: c014c406 00000007 d6b4c000 00000000 d6b4dfa4 d6b4df80 c016e0d8 c016d94c
[ 5702.852174] df80: 0000d004 b6de4654 b30fecc4 000afc80 00000036 c000ff24 00000000 d6b4dfa8
[ 5702.857677] dfa0: c000fd80 c016e0a0 b6de4654 b30fecc4 294b24c8 c014c406 b30fecc4 0000d004
[ 5702.863389] dfc0: b6de4654 b30fecc4 000afc80 00000036 00000020 b27095e8 b6de4504 b6dd4104
[ 5702.869072] dfe0: b6de4490 b30fecb4 b6dd2c94 b57fe17c a0000010 00000007 00000000 00000000
[... cut ...]
[ 5702.923194] [<c0520514>] (vchiq_prepare_bulk_data) from [<c05187ec>] (vchiq_bulk_transfer+0x2d8/0x554)
[ 5702.928991] [<c05187ec>] (vchiq_bulk_transfer) from [<c051d4f0>] (vchiq_ioctl+0x554/0x1958)
[ 5702.934610] [<c051d4f0>] (vchiq_ioctl) from [<c016d9dc>] (do_vfs_ioctl+0x9c/0x754)
[ 5702.940850] [<c016d9dc>] (do_vfs_ioctl) from [<c016e0d8>] (SyS_ioctl+0x44/0x6c)
[ 5702.943709] [<c016e0d8>] (SyS_ioctl) from [<c000fd80>] (ret_fast_syscall+0x0/0x28)
[ 5702.950478] Code: e51b2048 e5814004 e5841000 e5823000 (e5832004) 
[ 5702.953622] ---[ end trace d471ad6768c27035 ]---

Mike-Purcell, 2019-08-05:
Thanks Jason, sending Pi Zero back this morning.

jasaw, 2019-08-06:
Your kernel oops may be caused by the new Pi Zero W which has either bad components or bad design: https://github.com/raspberrypi/linux/issues/2555#issuecomment-394105250
The recommendation is to underclock it to 950MHz.

Mike-Purcell, 2019-08-07:
Thanks Jasaw. Got the new RPi Zero yesterday and at 0900 GMT this morning it has done exactly the same as the previous Pi Zero. I will try to under clock it now, post a reboot.

Mike-Purcell, 2019-08-09:
Setting RPis zero to clock at 950 seems to have done the trick and hasn't crashed in her 48 hours. Thank you for the help Jasaw.

closed 2019-08-09 by Mike-Purcell

jasaw, 2019-08-12:
@ccrisan Looks like the new batch of Pi Zeroes are not able to run at default clock speed. Do you know whether Raspbian has done anything to address this? Otherwise, we could just set our default to 950MHz?

ccrisan, 2019-08-12:
@jasaw as far as I know, they don't do anything about clock speed or overclocking in general, by default, in Raspbian, regardless of the board model.

Even though `config.txt` has a commented `arm_freq` that states a default of `700`, I believe that's actually chosen by the boot loader depending on the selected board.

motionEyeOS simply uses the very same boot loader, firmware, kernel as Raspbian does, the difference probably doesn't come from there.

Recently discovered that Raspbian runs the `rngd` service, which helps feeding the kernel with entropy for generating random data (or something like that). I added it to recent motionEyeOS builds and it seems to me that connecting to WiFi became a lot faster.

Now I haven't done too much testing with this `rngd` service and I can't tell if there are any downsides that come from using it, but I'd recommend trying recent nightly builds on the Zero, with the default clock, to see if the root problem goes away. How is that related to the CPU clock? I have no clue :)