motioneye-project/motioneyeos #2030: After upgrading to Chrome Version 76.0.3809.100, I can no longer see video from Motioneye

helmar74, 2019-08-08:
Hi,

I have several  motioneyeos cams running. 
After Upgrading to Chrome Version 76.0.3809.100 it's not possible to view the video live stream on the startscreen of motioneyeos.
Any other Browser (and also Chrome version 75.x.x.x) is working.
Any  idea how I get it running with chrome > 76.0.3809.100 ?

Thank's a lot

helmar

grhangelone, 2019-08-08:
Same here.
I'm using above version, too on my desktop linux boxes.
But I can take snapshots and watch them on the snapshot browser.

Looking at the pagesource, it seems to be stuck at the initializing.

[image]

I also just upgraded motioneye to the latest version because of this.
Doesn't help.

woznyp, 2019-08-08:
Hi, I had the same issue and as a workaround I've edited JavaScript files related to refreshing the video stream.
Please follow motioneye web interface repository thread:
https://github.com/ccrisan/motioneye/issues/1401

Nameback, 2019-08-08:
Same issue here, ended up rebuilding my whole motioneye setup only to find this thread. Didn't even think to check here. Good catch!

helmar74, 2019-08-08:
Hi,

here are the necessary commands:
mount -o remount,rw /
cd /usr/lib/python2.7/site-packages/motioneye/static/js/
cp frame.js frame.js.orig
cp main.js main.js.orig

Than copy the two file from rhe repository above to usr/lib/python2.7/site-packages/motioneye/static/js/

Than reboot motioneyeos

Than delete cache of Chrome.

That's all ;-)

helmar74, 2019-08-10:
Hi,
now I the live stream of the cameras are shown in Webinterface of chrome... but the left frame is greyed out and I can't click anything. It's the same on every browser.... reboot didn't helped....
[image]

ccrisan, 2019-08-11:
Already fixed in motionEye. Will be available in the currently baking nightly build.

closed 2019-08-11 by ccrisan