motioneye-project/motioneyeos #2031: Randomly stops showing camera feeds on android in chrome

talormanda, 2019-08-08:
Using chrome version 76.0.3809.89 on android.
Using motioneyeos on my raspberry pi.

The camera feeds do not load when I try to view them. I have tried clearing cache and site data and restarting my device but nothing has worked. I can view them on firefox on the same device, and they work fine when viewing from a desktop computer. Not sure why this randomly happened. Nothing was recently done to the OS at all.

[image]

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3480911/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3480912/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3480913/boot.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3480914/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3480915/motioneye.log)

PhinPhin, 2019-08-09:
I have the same issue. In Windows10 in Chrome I see the above spinning aperture-logo or (after some time) just the window in it's dark-grey backgroundcolor. It doesn't work in the incognito-window neither. CTRL+F5 or CTRL+SHIFT+F5 didn't do anything too. 

If I use Firefox it works suddenly again. Funny is yesterday evening I could still watch the streams, the  logs show nothing obvious.... the only thing that is unusual is dmesg.log:

`[  806.137484] uvcvideo: Failed to resubmit video URB (-1).`

Not sure if this is related but I bet this issue is the same?: https://github.com/ccrisan/motioneyeos/issues/2033

[messages (1).log](https://github.com/ccrisan/motioneyeos/files/3485213/messages.1.log)
[dmesg (1).log](https://github.com/ccrisan/motioneyeos/files/3485214/dmesg.1.log)
[boot (1).log](https://github.com/ccrisan/motioneyeos/files/3485215/boot.1.log)
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/3485216/motioneye.1.log)
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/3485217/motion.1.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3485218/boot.log)

BrotherBee, 2019-08-09:
Same problem here, works fine in Explorer but not in Chrome.
My Chrome: Version 76.0.3809.100 (Officiell version) (64 bit).
Some update for Chrome this night ?

FloRup, 2019-08-09:
I think this is the same problem as here: https://github.com/ccrisan/motioneye/issues/1401

Someone found a workaround and it worked for me
https://github.com/ccrisan/motioneye/issues/1401#issuecomment-519669086

PhinPhin, 2019-08-09:
Yes, setting chrome://flags/#enable-lazy-image-loading to disabled solved the issue. Duh, sudden Google Chrome-update......

ADYoungberg, 2019-08-11:
> Yes, setting chrome://flags/#enable-lazy-image-loading to disabled solved the issue. Duh, sudden Google Chrome-update......

Thanks for posting this! This fixed the issue for me. Thought something was wrong with my setup for about a week...

ccrisan, 2019-08-11:
Already fixed in motionEye. Will be available in the currently baking nightly build.

closed 2019-08-11 by ccrisan

talormanda, 2019-08-11:
@ccrisan does that mean we can auto update from the OS? or do we have to do something else to fix this ourselves if we are on motioneye V 0.40?