motioneye-project/motioneyeos #2032: Camera turns all white (stops working), else it is more purple than before (NOiR)

patrickhener, 2019-08-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20190728

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspi Noir v2

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet
### Peripherals

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/3481613/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3481614/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3481615/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3481616/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3481617/motioneye.log)

---

### Description of Problem

So the last update brought my camera to be more purple (cause of no infrared filter) which is not that bad. But at night it is bright white and you cannot see anything anymore. That was not there before.

Also it randomly changes to a white video output not showing anything anymore. Only a reboot can then reset the camera.

Also recently, but just once, the image froze. So it was showing the front door as it were normal but it did not change anymore. Also no motion detection and so on.

If you need further information, debug or logs just let me know.

Best regards,
and thanks in advance for support,
Patrick

jasaw, 2019-08-09:
I believe this was fixed by this PR: https://github.com/ccrisan/motioneyeos/pull/2021 but I don't think there's a dev build that includes this fix yet.

patrickhener, 2019-08-09:
If I understand the PR right it was merged to dev 9 days ago. As I use the dev branch I suggest it was not in the last release yet?

ccrisan, 2019-08-11:
The nightly build that's currently baking is going to have that commit. Our building machine was down for a while.

closed 2019-08-11 by ccrisan

patrickhener, 2019-08-13:
Unfortunately the nightly build still leaves me with a mostly purple cam feed and does not fix the problem.

patrickhener, 2019-08-16:
Is there a Button to reopen? @ccrisan?

popoviciri, 2019-08-16:
Hi @patrickhener, you're not the only one with this problem. For sure there are plenty of NoIR cameras out there, the main artifact for a surveillance system. The latest kernel updates and subsequent firmware have changed the Auto White Balance (AWB) mode for NoIR cameras. To get a first read about it, you may check this: https://github.com/raspberrypi/firmware/issues/1167 or search "AWB" and "NoIR" through https://github.com/raspberrypi.
My point is: this has nothing to do with motioneyeos or its developer @ccrisan. It is coming from upstream and will take some time to have everything sorted at the ends. See PR https://github.com/ccrisan/thingos/pull/28 for example; and there are others.
- If you need something on short term, I suggest you go back to the release that worked for you in the past. I believe the June 19 was still before things changed.
- Or try this one: https://github.com/popoviciri/motioneyeos/releases/tag/dev20190816r. If you using this image with motion, you must put this in the "Extra Motion Options" box: `mmalcam_control_params -awb greyworld`. If you use it as Fast Network Camera (FNC), the AWB option Greyworld will not work with RTSP, but only with mjpeg streaming.
- A third option, would be to wait and use the images as are now. It is purple, but not as bad as few releases back.
Cheers!

patrickhener, 2019-08-26:
Hello @popoviciri . And thank you for the really detailed explanation. I can understand what you said and starting motion with the extra options worked like a charm. Thanks again and have a nice week.