motioneye-project/motioneyeos #2033: Unable to view feed from MEYE web GUI.

RyGuy0248, 2019-08-08:
I am running motionEyeOS version: dev20190728.

I am using the following board/model: Raspberry Pi Zero W.

I am using the following type of camera: V4L2.

My camera model is: Logitech C615.

My motionEyeOS unit is connected to the network via: WiFi.

I consider the following log files relevant to this issue:

motioneye.log - https://pastebin.com/JJB6Ub88
boot.log - https://pastebin.com/T5RHRKns

I am unable to view my camera feed by visiting the web GUI. If I went to the snapshot or streaming URL, I can view it fine.

TouristInformation, 2019-08-09:
I guess I have the same Problem. I'm using the original Raspberry Foundation Camera Module V2 and dev20190728. All I see is arotating circle in GUI, but when I use raspistill in the shell I can take photos. So the camera seems to be ok.

PhinPhin, 2019-08-09:
As seen above I referenced this in another thread, having the same issues. Funny thing is, it just stopped working last night it seems. Cannot see why.... reboot doesn't help. If I click the snapshot-button in the webinterface it takes one normally and I can view it in the webinterface, too.

JorisBry, 2019-08-09:
Works in chromium and in firefox, not in Chorm on Win10 or android.
I guess it is related to the way chrome handles the javascript, but I'm not knowledgable enought to figure it out

PhinPhin, 2019-08-09:
I wonder why it suddenly happend here...... it won't pull any updates automatically, aside from NTP, right?

PhinPhin, 2019-08-09:
As it was stated in https://github.com/ccrisan/motioneye/issues/1401 it seems to be a bug (that is already fixed in the devbranch) that is triggered by the latest Google Chrome update.

Workaround: go to chrome://flags/#enable-lazy-image-loading and set to disable.  Check if this helps, @RyGuy0248

JorisBry, 2019-08-09:
workaround works. I'll update to the newest release once avalable. Thanks

RyGuy0248, 2019-08-09:
> As it was stated in [ccrisan/motioneye#1401](https://github.com/ccrisan/motioneye/issues/1401) it seems to be a bug (that is already fixed in the devbranch) that is triggered by the latest Google Chrome update.
> 
> Workaround: go to chrome://flags/#enable-lazy-image-loading and set to disable. Check if this helps, @RyGuy0248

This works! Thank you!

closed 2019-08-09 by RyGuy0248