motioneye-project/motioneyeos #2034: All cameras stopped working

bobster316, 2019-08-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 4.

### Camera

Wyze Cam v2

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g.  Ethernet.

[image]
[motion.log](https://github.com/ccrisan/motioneyeos/files/3488084/motion.log)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

bobster316, 2019-08-09:
When checking with VLC all Wyze camera's work fine

Weeman89, 2019-08-09:
Just signed up to github to leave a comment.

I've had the same/similar problem, are you using Chrome and have you tried Firefox?

bobster316, 2019-08-09:
Currently using motioneyeos-raspberrypi4-dev20190715

bobster316, 2019-08-09:
Just tried with Firefox and it works! Thanks
what is stopping chrome from running it?

SoloSquirrel, 2019-08-10:
Same issue, all cams stuck in spinney-circle mode.  Tried Firefox, and everything works fine.

helmar74, 2019-08-10:
look at #2030

grhangelone, 2019-08-10:
Go to chrome://flags and search vor "lazy"
disable both options
for now this helps the most

helmar74, 2019-08-10:
Hi, I disaled both options, but it's not working for me....
[image]

ccrisan, 2019-08-11:
Already fixed in motionEye. Will be available in the currently baking nightly build.

closed 2019-08-11 by ccrisan

neilyoung, 2019-08-12:
@ccrisan When will the fix be available in the raspberry PI 3 images? Is there any way to get notified?

ccrisan, 2019-08-23:
@neilyoung use the latest nightly build.