motioneye-project/motioneyeos #2044: Help with firewall configuration

tastewar, 2019-08-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

Raspberry Pi Zero W

### Camera

I am using the following type of camera: Simple MJPEG Camera

My camera model is: RasPi camera module

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

none

### Log Files

none

I had everything working previously with 201806?? but then yesterday for some reason, instead of the video feed, I was just seeing a spinning iris.  I tried rebooting, but still the same symptom. So what the heck, I thought, I'd try updating. Tried the auto update, and only later discovered that I shouldn't have done that. Oh well, so I started from scratch with the latest supported version. After a couple of false starts due to my failure to configure wifi ahead of time, I finally got it up and running. It is currently all working fine from inside my network. I have a registered DDNS name, and have my router configured to forward port xxxx to port 80 on the eyepi host. I have not changed my router settings from before! However, it's not working from outside my network. The basic UI shows up, but instead of my video feed, I am getting a camera with a line through it. The FAQ suggests I need to perhaps forward (something) to port 8081 on the pi.

So my confusion stems from the fact that previously, I only had the one port forwarded. Which suggests to me that I have something different in the settings now from before. Who wants to guess what I have misconfigured? I mean, seriously, it had been working flawlessly from outside my network since I had installed it last year, and with only that one rule.

So... currently, the most suspicious thing I can think of is that I have FastNetworkCamera enabled. But prior to that, I wasn't getting an image even locally.

Thanks for any clues!

tastewar, 2019-08-18:
OK, hearing nothing then, I added a second firewall rule on my router to direct incoming traffic on the streaming port to the eyepi unit. But I didn't delete such a rule before, so I believe it was all working with one port forwarding rule previously. Confused.

DuMaM, 2019-08-27:
> I had everything working previously with 201806?? but then yesterday for some reason, instead of the video feed, I was just seeing a spinning iris. I tried rebooting, but still the same symptom. So [...]

Hi,
I think yours problems was connected with some errors discovered on newest Google Chrome. I got similar issues even though later it was working nicely. So when you are setting up everything please check it also on other browser (firefox). 

If i good understand you got such setup:
local network camera -> router -> DDNS service -> router -> local rpi with motioneyeos -> outside network -> your pc 

So the best option is to check everything step by step. :]

1. Check if you can see your local camera stream via any player(VLC can be good) or https/http. 
yes -> Got to 2 step. 
no -> Fix local connection on your camera.
2. Setup ddns:
a) If your firewall do not pass ddns service, do it manually and forward ports 53:53.
b) you has to forward camera_stream:port for this service too, because ddns will be unable to see other ports as 80 and 443. Both input and output.
c) go to ddns provider and set it

Check it according to [this site](https://www.siteground.com/kb/check-port-blocked-isp/)

Does your ddns service work as it should?
yes -> Got to step 3
no -> Verify step a and b and then c :unamused: 

3. Setup rpi locally.
Add camera via ddns address it should work because you earlier forward input and output traffic.
Do you see working camera from local motioneye?
yes -> Got to step 4
no -> Check steps 1,2,3 :disappointed: 

4. Setup rpi remotely
By default rpi is listening on 80 with motioneye frontend so to make it work ass supposed to you, has to forward this port on any other port that you want to use for example 8080.
Next get your [outside ip](https://www.iplocation.net/find-ip-address) and check if you can see your site on:
https://outsiteip:selected_port
Does it work now?
yes -> Great :tada: 
no -> Check steps 1,2,3,4 :sob: 

I hope this would help :pray: 

BR

ccrisan, 2019-08-29:
Google Chrome stream refresh bug. Fixed in recent nightly builds.

closed 2019-08-29 by ccrisan