motioneye-project/motioneyeos #2045: Opening motioneyeos on different devices

SimonLynch1962, 2019-08-14:
I have set up motioneyeos on a pi zero and it boots up successfully ... however I can only view the camera on my mobile phone. If I try either my tablet or laptop, using the same url, I don't see the stream from my camera ... just a spinning progress icon. Any ideas why?

jasaw, 2019-08-15:
which web browser did you use? Also try the latest nightly dev build?

SimonLynch1962, 2019-08-15:
I am using Google Chrome under Windows 10 on the Laptop and Chrome for Android on the tablet. Chrome for Android works with my phone but is probably a newer version/revision.

When I downloaded the Motioeyeos image, I just took the latest version for Pi Zero

raspberrycoulis, 2019-08-15:
I suspect this may be linked to the issue I just created [here](https://github.com/ccrisan/motioneyeos/issues/2047)