motioneye-project/motioneyeos #2047: Accessing web viewer does not stream footage in Chrome

raspberrycoulis, 2019-08-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427
--

### Board Model

I am using the following board/model: Raspberry Pi 2 (model B)

### Camera

I am using the following type of camera: MMAL

My camera model is: Raspberry Pi Camera Module (v1)

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB WiFi adapter

### Log Files

I consider the following log files relevant to this issue:

 * [motioneye.log](https://github.com/ccrisan/motioneyeos/files/3505481/motioneye.log)
 * [boot.log](https://github.com/ccrisan/motioneyeos/files/3505482/boot.log)

---

I have been using motionEyeOS for a number of years now - everything has worked fine, including updating and accessing my camera over HTTPS (via a reverse proxy) - this all worked fine, and continues to.

However, recently I have not been able to view my camera in Chrome (version 76.0.3809.100 (Official Build) (64-bit)) - the web UI loads normally, but where the camera footage should be just stays grey.

I can confirm that I can still access my camera feed in Safari, Firefox and Edge without issue (both over the HTTPS custom domain via reverse proxy and directly via the IP address when accessing on my LAN / VPN), which seems to indicate a possible issue with Chrome.

Nothing in my setup has changed over the last 2+ years. Pi works fine, USB WiFi adapter all good and I have also tried rebooting the device just in case - the issue still remains.

raspberrycoulis, 2019-08-15:
The following issue seems to highlight this in a follow-up comment:

> I am using Google Chrome under Windows 10 on the Laptop and Chrome for Android on the tablet. Chrome for Android works with my phone but is probably a newer version/revision.
> 
> When I downloaded the Motioeyeos image, I just took the latest version for Pi Zero
> 
_Originally posted by @SimonLynch1962 in https://github.com/ccrisan/motioneyeos/issues/2045#issuecomment-521555740_

SuneBielefeldt, 2019-08-15:
Same issues here ...
First, my MotionEye cameras stopped working on my Samsung Galaxy with Google Chrome. Back then, it worked on both my PC and my Samsung Galaxy Tab.
But then yesterday, Google Chrome on Windows 10 also stopped working.

I can still access MotionEye with Internet Explorer on my PC.
In addition, I can access MotionEye on my Android phone through Samsung's browser. But Google Chrome, has stopped working.
Rotating camera images only.

Google Chrome on Windows 10
[image]

Google Chrome on Internet Explorer
[image]

raspberrycoulis, 2019-08-15:
@SuneBielefeldt can you post the version of Chrome too, may be helpful. You can go to chrome://version/ in your Chrome browser to get the detail

judalvarezca, 2019-08-15:
Same problem here, I'm currently having issues with Chrome and Safari, the frames keep on loading GIF and never connects to the feed.

SuneBielefeldt, 2019-08-15:
I'm using Google Chrome browser version 76.0.3809.100.

[image]

ccrisan, 2019-08-15:
Fixed in recent nightly builds.

closed 2019-08-15 by ccrisan

SuneBielefeldt, 2019-08-15:
Hi Calin Crisan
You are a super man to solve such problems so quickly :-)

judalvarezca, 2019-08-16:
Already installed the 0.41rc1 build! Thank you very much for the quick fix and response :+1:

SimonLynch1962, 2019-08-16:
Can anyone provide a link to the the latest update (and does it work on Pi Zero) ... I need to refresh my SD card and try again

judalvarezca, 2019-08-16:
https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev

raspberrycoulis, 2019-09-05:
Appreciate this may be resolved in a nightly build, but is there any ETA on when this will be part of a stable release? Would much prefer to use the update mechanism in the UI than essentially reflashing an SD card with the nightly build...