motioneye-project/motioneyeos #2051: Frequent hanging and freezing.

postal302, 2019-08-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 4.2.2 20190427.

### Board Model

I am using the following board/model: Raspberry Pi Zero W.

### Camera

I am using the following type of camera: MMAL, Network Camera.

My camera model is: Pi Camera Model v1.3.

### Network Connection

My motionEyeOS unit is connected to the network via WiFi.

### Log Files

Here's the logs I'm working with:
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3508130/motioneye.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3508131/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3508132/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3508141/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3508142/motion.log)

Hey!

First let me say this is a great thing you've built here! I also very much appreciate all the work you constantly put in! I've been scouring the posts and Google and haven't seen the silver bullet to an issue I'm having.

I have multiple Pi Zero Ws running the current version of MotionEyeOS, but one keeps hanging and crashing. I've reviewed the logs and while I can find many errors, I can't seem to pinpoint what causes the problem. I've made many changes to my configuration including resolution, movie quality, video format and more. I've done all I can and can't seem to fix the problem.

Previously I had a different Pi Zero W using this specific camera with the same MotionEyeOS version and there were no issues, and the server continues running even after the image stream stops; so I feel like I can rule out hardware faults. Also there isn't any WiFi issues.

I'm at a loss and open to any insight you can give!

Thanks,
Phill

jasaw, 2019-08-16:
The instability is probably caused by the new Pi Zero W hardware. See https://github.com/ccrisan/motioneyeos/issues/2025
Try the latest nightly build first. If that doesn't work, try underclocking it to 950MHz.

postal302, 2019-08-16:
Man you guys are on it!! Thank you @jasaw for the quick response! I updated to the latest nightly and it has now been running for ~10 hours without issue (which is way better than the previous 1 to 1.5 hours). This appears to be the fix. I’ll comment again if I see an issue.

closed 2019-08-18 by postal302