motioneye-project/motioneyeos #2054: UV-IR CUT filter

freddy64, 2019-08-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 Hello, Can an IR-CUT Filter that can be put in front of the lens of a Raspberry Pi NOIR camera be replaced by a program that would feature in the settings of motioneysOS. Or is it just not possible to replace a physical filter ?

freddy64, 2019-08-18:
Hello, Can an IR-CUT Filter that can be put in front of the lens of a Raspberry Pi NOIR camera be replaced by a program that would feature in the settings of motioneysOS. Or is it just not possible to replace a physical filter ?

jasaw, 2019-08-19:
If there's a lightweight program that we can include into MotionEyeOS image, ya sure, why not. Do you know a program that can do this?

freddy64, 2019-08-19:
I'm not a programmer and I don't know if such a program exists. If an IR-Cut filter was included in the motioeyesOS settings, I could view my Raspberry Pi NoIR during the day with normal colours. Associated with a working schedule it could then be switched off at night and benefit from the infrared Camera Module v2.

jasaw, 2019-08-20:
I did a bit of reading on this topic. Without the IR-cut filter, the blue sensor is swamped with IR, and there's no way for software to workout the correct level of blue. Those cameras that feature software night vision do it differently. They have IR-cut filter on the camera, but for night vision, they combine the RGB output to get white colour, then use software to turn the brightness up. This has the drawback of noisier image at night.

If you have NoIR camera, but don't want to have an IR-cut filter, you could try setting the auto-white balance (awb) to tungsten.

freddy64, 2019-08-21:
Thank you for your reply and explaining why you can't replace an IR-Cut filter simply with a software program. I couldn't find the auto-white balance settings in the GUI motioneyeOS settings and I don't know where to find them in the terminal through ssh. I did find a Raspberry Pi IR-CUT Camera with Night Vision at Aliexpr.... They claim it works like a normal camera during the day and as a NoIR at night.

jasaw, 2019-08-22:
Try adding `mmalcam_control_params -awb tungsten` to "Extra Motion Options"?

If you're buying RPi camera with IR-cut filter, be aware that there are 2 types of cameras available:
1. Automatic IR-cut filter control: The filter is controlled automatically by the light sensor on the camera PCB. There is no way for software to override the control, but really easy to use.
2. IR-cut filter controlled via camera LED: This requires additional software to set the camera LED on/off which actually controls the filter on/off.

freddy64, 2019-08-24:
Hi Jasaw, I added that command as you suggested and it's turned out quite well, thanks.

I didn't find an Automatic IR-cut filter from my raspberry pi dealer(Kubii), so I bought one from Aliexpr..It's a 5MP and compared to my 8MP v2 camera module and costs less. I will see how it performs.

popoviciri, 2019-08-26:
Hi @freddy64, try the latest dev20190820 build and add `mmalcam_control_params -awb greyworld` to "Extra Motion Options".

freddy64, 2019-09-03:
Hi popoviciri, I downloaded the latest build and added -awb greyworld. The image is still quite reddish and the -awb tungsten works better for me. Where do you find all of these commands? I could try them all out and see which one I prefer.

popoviciri, 2019-09-03:
Hi @freddy64, make sure your image is indeed updated. I just tried the latest dev (released today) of my cams and it works fine. This is the image that I just flashed onto:
[image]
Try to re-flash the SD instead of upgrading from the webui.
Check also for any typos in your command: `mmalcam_control_params -awb greyworld`. Other AWB options here:
[image]
But you may pass any of the camera commands via `mmalcam_control_params`. A ful list here: https://www.raspberrypi.org/documentation/raspbian/applications/camera.md
Cheers!

freddy64, 2019-09-03:
Hi @popoviciri, I re-flashed the SD card today to dev20190903 and added the -awb greyworld command and it's come out really well, thank you very much.

Incidentally, removing the SD card from the RP zero case is quite fiddly and I'm always worried about tearing the short ribbon cable. I might just cut out a slot in the back of the case and with a pair of tweezers be able to remove the SD card without having to open the case every time I need to re-flash the SD card.. Just a thought.