motioneye-project/motioneyeos #2056: No camera image is displayed

gezuka77, 2019-08-18:
Hi,

For some time I have this strange issue. When looking the cameras in the chrome browser, I only see the "spinning loading signs": [https://ibb.co/Xbz3dJD](url). It seems to work with edge.

The recording works as I can watch back the clips.

Here [https://community.brave.com/t/motioneye-camera-streaming-not-working/72005](url), i see the same issue reported.

Ideas?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

gezuka77, 2019-08-18:
Never mind. Found the fix here [https://github.com/ccrisan/motioneye/issues/1401](url)

ccrisan, 2019-08-23:
Fixed in recent nightly builds.

closed 2019-08-23 by ccrisan