motioneye-project/motioneyeos #2057: No video Preview in admin panel

samuelcomplex, 2019-08-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model:Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2

My camera model is: Logitech C310

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi 5Ghz

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:
[image]

samuelcomplex, 2019-08-19:
Stream works when going directly to x.x.x.x:8081
Recording and snapshots work as well.
Have dried changing streaming auth to basic and using password but that did nothing.
Both dimmers are set above 1.

chunkysteveo, 2019-08-19:
I can also confirm this issue - thought I was going mad. No video preview from the camera in the admin panel. I can view the stream if I enable it on port 8081, and the motion detection and video saving (to Google Drive) works fine - it all works as normal and as expected - just no video in the admin panel for the camera?

chunkysteveo, 2019-08-19:
Ahh - following another issue raised and looking at closed issues - Does not work in Chrome, works in Firefox as a test. Happy to wait till the release candidate becomes the next stable release - glad it's not a bug of MotionEyeOS.

@samuelcomplex - have you looked at closed issue https://github.com/ccrisan/motioneyeos/issues/2047 ? Try the Web admin in another browser?

samuelcomplex, 2019-08-19:
> Ahh - following another issue raised and looking at closed issues - Does not work in Chrome, works in Firefox as a test. Happy to wait till the release candidate becomes the next stable release - [...]

Can confirm it works in edge, just not chrome. Did not see this other issue even though I searched. Thank you!

You mentioned the next release candidate, does that have chrome support?

Gomtu, 2019-08-19:
Same goes here. IE 11 works too. Just not chrome.

chunkysteveo, 2019-08-20:
@samuelcomplex If you go to the releases page (https://github.com/ccrisan/motioneyeos/releases) you can see a "nightly build", which is a testing and ongoing version full of the latest bug fixes, tests etc. A Pre-release (20190619) which contains a lot of new fixes and features, but not yet confirmed stable, and the current stable release (20190427) which is the latest stable and "correct" version. (plus other older versions after that!) 

From closed issue https://github.com/ccrisan/motioneyeos/issues/2047 looks like @ccrisan fixed the Chrome issue in the nightly builds which you can try out instead of the stable version. Or maybe ask him when the fix will be added to the stable release. It won't be in the pre-release (20190619) as the fix was only added to the nightly builds a few days ago.

Steve

samuelcomplex, 2019-08-20:
@chunkysteveo

Thank you. 
20190619 loads indefinitely in the webgui but the latest nightly build works, apart from wifi, so I'll use that for now.

Thank you again for the help.