motioneye-project/motioneyeos #2058: On / OFF live video on demand

dimgod59, 2019-08-20:
Preliminary Docs
I confirm that I have read the CONTRIBUTING guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version
I am running motionEyeOS version: motionEyeOS 20190427.

Board Model
I am using the following board/model: Raspberry zero w.

Camera
I am using the following type of camera: Local V4L2 Camera.
My camera model is: camera generic raspberry.

Network Connection

My motionEyeOS unit is connected to the network via: wifi

Hello,
can you explain to me (if it is possible) how to give a command (URL? API? SSH?) which makes it possible to stop the diffusion of the camera? for example when one is present one stops the camera (live video) and when one is absent one activates the camera (live video)

Best.

Dim

jasaw, 2019-08-22:
See https://github.com/ccrisan/motioneye/issues/203#issuecomment-220272592

dimgod59, 2019-08-26:
Good evening,

I am a great debutant in the linux / ssh world. Excuse me in advance :)
If I understood correctly on the different pages, I would like to make ON / OFF streaming on request for example via an SSH command.

To do this:

I created 2 files:

mkdir StreamingON: I write: systemctl start motioneye.service
mkdir StreamingOFF: I write: systemctl stop motioneye.service

then I just have to do in the terminal:

Sudo ssh admin @ IPmotioneye

 sudo StreamingON;
or
 sudo Streaming OFF

should it work?

Best

dimgod59, 2019-08-31:
the link it's for MotionEye, it's impossible for motionEyeOS ?