motioneye-project/motioneyeos #2059: How to figure out what my EXTERNAL IP Address is?

Goldglv, 2019-08-20:
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

I am running motionEyeOS version: 0.41rc1

I am using the following board/model:Raspberry PI 3B+

I am using the following type of camera: MMAL

My camera model is: Raspberry Pi Camera V2

My motionEyeOS unit is connected to the network via: WiFi

Since there is no web browser in MotionEye (normally I would just go to https://whatismyipaddress.com/), how do I figure out what my EXTERNAL IP Address is so I can remotely watch the stream OUTSIDE my local network?

jasaw, 2019-08-21:
Since you're on Wifi, you should be able to visit whatismyipaddress website from any computer within your local network. All your devices on your local network will have the same public IP address.

If you want to access your motionEyeOS device from the internet, you'll need dynamic DNS configured on your router if your ISP put you on DHCP. You also need to enable port forwarding on your router to forward to your motionEyeOS device. I'm not sure which parts you need to forward.

Goldglv, 2019-08-21:
All of my devices will have same public IP?  Wouldnt they all have diff ip's???

jasaw, 2019-08-21:
Your ISP assigns one IP address to your router. That is your public IP address. All your devices will have different local network IP address that is leased out by your router (if you have DHCP enabled).

Goldglv, 2019-08-21:
Great, thank you for explaining!  What if I have more than one Raspberry Pi with a camera and I wanted to live stream both cameras?  If there's only 1 public IP, which feed would I see?

Chiny91, 2019-08-21:
To see one of your cams, you will have find the "port forwarding" web (admin) page of your router and re-direct http:// requests made to your public IP address over to your R Pi.

Maybe router dependent, not sure, but years ago, I was able to access 2 different R Pi by adding a port number to my single public IP address, as:
[public IP address]:4000 re-directing to [R Pi 1 IP address]:motionEyeOS port
[public IP address]:4001 re-directing to [R Pi 2 IP address]:motionEyeOS port

Although many people are port forwarding from the big bad internet to their humble R Pi, it must be considered a bit risky.; less risky for motionEyeOS as it is a stripped down OS under active development but still not ideal.   Far better, would be to run an OpenVPN server on your network; again, a feature provided by some routers.

gitgoldberg, 2019-08-22:
I use realVNC to access my Raspberry Pis from outside my home network.  This method is reliable, secure, and does not require port forwarding to be set up on your router.  RealVNC supports Raspberry Pi and allows free access to up to five Pis.

Goldglv, 2019-08-22:
@gitgoldberg will this work with Motioneye?  Wondering since Motioneye is it's own OS, nothing else gets installed other than Motioneye.

Goldglv, 2019-08-22:
I just saw author mentioned RealVNC will NOT work with Motioneye because of what is stated above....guess it's back to port forwarding.....

gitgoldberg, 2019-08-22:
Goldglv
You're right, my mistake, apologies.
RealVNC will only work with motioneye over Raspbian, **not** motioneyeos.
I had forgotten that after several years of using motioneyeos I switched to motioneye over Raspbian earlier this year so that I could use RealVNC and also monitor devices such as ds18b20 on the gpio.

closed 2019-08-23 by ccrisan