motioneye-project/motioneyeos #2060: starting wpa_supplicant: failed (brcmfmac)

Mapamann, 2019-08-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

motionEye Version | 0.41rc1
Motion Version | 4.2.2
OS Version | motionEyeOS dev20190820

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

Raspberry PI 3B+
I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

MMAL Camera
I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hello,

I tried everything what came to my mind to try to get my PI 3+ to connect to my wireless network. 

With the latest stable release 20190427 it connects to my wifi (tried 5Ghz and 2.4Ghz)
But with the last two pre-releases and the same wpa_supplicant.conf that works on 20190427  i get the  starting wpa_supplicant: failed (brcmfmac) message on startup.

When i connect the cable and log in to the web interface i even can see under the wireless settings that it filled the SSID and key out of my wpa_supplicant.conf but it will not connect to the Wifi on startup.

I even set up a Raspean and did copy the wpa_supplicant.conf from there but same result.

Can anybody help me how to fix this?

Mapamann, 2019-08-23:
Nobody any idea?

jarakangas, 2019-09-08:
Same problem here. Ethernet is OK. But with wifi I also get "failed (brcfmac)" error.

barneycoles, 2019-09-09:
Same problem too.  Ethernet is OK. But with wifi I also get "failed (brcfmac)" error.

CoolHandMike, 2019-09-28:
I had the same issue with the same Pi. This is how I fixed it:

1. Load fresh image (20190911) onto the micro SD card with Balena Etcher.
2. Open notepad++. Set to UNIX mode: Edit-->EOL Conversion-->UNIX (LF)
3. This is what's in my wpa_supplicant file (yes, keep the quotation marks):

country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
key_mgmt=WPA-PSK
psk= "mywifipassword"
ssid= "mywifissid"
}

I had to confirm my scan_ssid channel by looking at my wifi config on my router. It's set to channel 1. I'm also using WPA2, so key_mgmt gets set to WPA-PSK.

4. In the notepad++ dropdown for "Save as type:" scroll up one and select All. Save it as wpa_supplicant.conf. I saved it to my desktop so I could copy/paste from there.
5. Open up your newly flashed micro SD card. Ignore Windows telling you it needs to be formatted. You should still be able to access the boot partition and see some files there.
6. Copy your wpa_supplicant file from your desktop and paste it into the SD card boot drive.
7. Use the taskbar to safely eject the SD card.
8. Install the card in your Pi and see if it works. It should say **Starting wpa_supplicant: done** if you were successful.

Unfortunately it may take some trial and error to get the settings just right. Each time you change something in the conf file, it's better to perform the whole procedure as I have outlined from start to finish since I think the wpa_supplicant file is only accessed during first boot, then becomes read only.

Good luck!

reinst, 2019-10-12:
If the above is still not working try this: My background, I used ethernet for the original setup and from other posts I read I needed to setup wifi pre-boot. I got wifi working anyways.

After the wifi was not working I got an error: _starting wpa_supplicant: failed (**brcmfmac**)_ <-- this will be different based on your wifi driver.

1. Pull the micro sd out and mount back on your regular computer:
2. Open the 231MB partition
2a. As admin - open /etc/wpa_supplicant.conf: edit
```
country=US
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    scan_ssid=1
    ssid="wifi_ssd_name" <- enter yours 
    psk="password" <- enter your pass
}
```
2b. **Save**

3. Open 2nd partition (should be the remaining size of your micro sd card)
[image]

3a. As admin - open /etc/wpa_supplicant.conf: edit

```
country=US
update_config=1
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    scan_ssid=1
    ssid="wifi_ssd_name" <- enter yours 
    psk="password" <- enter your pass
}
```
3b. **Save**

As you can see I did not save any wpa_supplicant.conf file in my main partition OR did I save one in my /boot (not shown)
[image]

Roozeppe, 2019-12-15:
Thanks, @reinst. This works for me.

danschimpf, 2020-12-15:
I tried all of the solutions mentioned above on my mac, with no success. 

boot up would see the file, then indicate "failed (brcfmac)" error. (I was changing the extension to .conf, otherwise during boot up would not indicate "Starting wpa_supplicant:   " and would try to boot a wired connection) 

Then I tried downloading, editing and saving the file on my PC (just my specific info added to the template no additional verbiage) ..and it worked 1st time.

perhaps something with the formate of a mac txt file was unreadable.  

on the PC, I saved the file as an ANSI file

not saying this is the solution, just adding what appears to have worked for me

LEONIDAS-0117, 2021-01-13:
I have being trying to connect to a 5GHz network that only admits WPA2 as Encryption method and CCMP as Cipher for the past few days. This is the wpa_supplicant.conf that worked for me:

```
country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant
        
network={
    scan_ssid=1
    ssid="network-name"
    psk="password"
    proto=RSN
    key_mgmt=WPA-PSK
    pairwise=CCMP
    group=CCMP
    auth_alg=OPEN
}
```
If not sure what Cipher is being used you can edit:
```
pairwise=CCMP
group=CCMP
```
to
```
pairwise=CCMP TKIP
group=CCMP TKIP
```

Normanras, 2021-03-15:
Tried setting up motioneyeOS a few days and failed and I was just able to connect it to a monitor and lo, the same issue. Anyone who solved with the above suggestions - has anyone used a mac? My PC is currently unavailable. 

I tried using Atom to convert the line-endings to Unix (LF), but it still didn't work. 

I'm also wondering why any other distro for RaspberryPi I've used, has worked without an issue. I use Text Edit, make the file with proper code, save it as wpa_supplicant.conf, add a second empty file called ssh and I'm in on first boot. Strange that motioneye specifically is the issue here!

starbasessd, 2021-03-15:
Can you post your wpa_supplicant.conf file here?
What Pi board are you using?
Which version of motionEyeOS are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RainNB08, 2023-05-04:
none of the solution above works for my raspberry pi 0 with the latest version of motion eye

starbasessd, 2023-05-04:
Which PiZero do you have?  On the back of the board, under the GPIO pins, it has 2 lines of text, Please post them here.

RainNB08, 2023-05-04:
PiZero W

starbasessd, 2023-05-04:
This was a link back to this issue...

RainNB08, 2023-05-05:
Is there any thing i can do now?

RainNB08, 2023-05-06:
[image]

starbasessd, 2023-05-06:
Use the [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz) image for PiZero
DO NOT USE Raspi-Imager, use Etcher or Win32DiskImager to create the SDCard.
Raspi-Imager will corrupt settings within motionEyeOS
Use a wpa_supplicant.conf with the format:
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="password"
}
- - - - -
Substitute your SSID and password for your network, and your country...

RainNB08, 2023-05-06:
it still failed
same error
failed(brcmfmac)

starbasessd, 2023-05-06:
Can you post the text of your wpa_supplicant.conf file here?
How did you create it and how do you copy it to the /boot partition before first boot?

RainNB08, 2023-05-06:
i copied the code you have sent directly into the wpa_supplicant.conf file, and fill in my own ssid and passcode. 
I just opened a new file my D drive along with overlay and other files

starbasessd, 2023-05-06:
Do you have "View Extensions" turned on in File Explorer? If not, it may be creating wpa_supplicant.conf.txt instead of wpaq_supplicant.conf
You should not be using Notepad to create the file. It is better to use Notepad++ or other Linux friendly text file editor.

RainNB08, 2023-05-06:
yes i have, i am sure that the wpa_supplicant.conf is correctly named and I am using the notepad++

starbasessd, 2023-05-06:
Do you have a 2nd PiZeroW to test with, or a 2nd SDCard? I've just tested 2 PiZeroWs and 3 SDCards, and all work interchangeably.

RainNB08, 2023-05-06:
i have tried two sd card, but on the same PiZeroW, is there anyother cause that might lead to this failure?

starbasessd, 2023-05-06:
Other than a defective PiZeroW, the only other thing(s) I can think of would be a wifi password or ssid with a disallowed character, or a wifi with a 5GHz name that is the same as a 2.4GHz name (11AC same as 11G or 11N) and it's trying to connect to the 5GHz access point instead of the 2.4GHz access point...

RainNB08, 2023-05-07:
I am positive that my PiZero W is not malfunctioning because the raspberry pi os run perfectly beside lagging, but hwo should i deal with the wifi situation

RainNB08, 2023-05-07:
also, somehow my /boot does not have loader.bin in it

RainNB08, 2023-05-08:
Thanks @starbasessd

AvidCoder1, 2024-01-05:
If anyone is still trying to find a solution that ACTUALLY works, I spent hours on this and found one:

I am using a RPI Zero W (2017) which only does 2.4ghz. My network has a shared SSID for both 2.4 and 5ghz. My service provider will not let me change this.

If you are in a similar situation, here is the fix:

Specify a frequency list in your "wpa_supplicant.conf" file (either in the global section or in the network section)

Example:

`ctrl_interface=DIR=/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

freq_list=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 
# Here ^

network={
    ssid="YourNetwork"
    psk="VerySecretPassword"
    freq_list=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472
    # OR Here ^
}`

starbasessd, 2024-01-05:
@avidcoder Interesting. I have several APs with the same ssid for both 2.4 GHz and 5 GHz. I have yet to have an issue. I have Pi1, (dongle), Pi2 (dongle), RPi3 (onboard & dongle) RPiZero (dongle & onboard), and RPi4 (onboard & dongle).

NEOBRETON, 2024-04-15:
I have the same problem:
On a Pi zero W, I install a Motioneye image (20201026), adding a "wpa_supplicant.conf" corresponding to one of my local networks, and everything works perfectly, but when (on the same configuration, I load the same image (20201026) with a "wpa_supplicant.conf", modified for another network, I'm given (as soon as I boot) the "brcmfmac" failure message. Note that everything is edited with Notepad++v8.4.8.
Any help would be appreciated.
Here are copies of the two files:
The one that works:
country=fr
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
ssid="SFR_1C78"
psk="26ptxt2kez54rjrm3fnb"
}
And the one that doesn't work:
country=fr
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
scan_ssid=1
ssid="RESEAU-2.4G"
psk="284Vi456"
}

starbasessd, 2024-04-15:
@NEOBRETON are you within range of the 2nd wifi AP when you start it up the first time?
You can try using this format in your wpa_supplicant.conf: (before first boot)
- - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="SCHOOLS NETWORK NAME"
    psk="SCHOOLS PASSWORD"
    id_str="school"
}

network={
    ssid="HOME NETWORK NAME"
    psk="HOME PASSWORD"
    id_str="home"
}
- - - - -

NEOBRETON, 2024-04-15:
Yes, I'm within range of both routers.
I tried your configuration suggestion, and the Pi always connects to the same router (the second, "home"). So I turned off its wifi, and at this point the Pi doesn't connect (brcmfmac).
The router with which the connection isn't being made isn't connected to my modem, could this be the reason?
It's 4:26 a.m. here, so I'll have a look tomorrow...

starbasessd, 2024-04-15:
@NEOBRETON Doubtful. If you only have the 2nd failing ("RESEAU-2.4G") then I would guess either the SSID or password is bad. If you install RPiOS Buster [(From here)](https://downloads.raspberrypi.com/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-05-03/2023-05-03-raspios-buster-armhf-lite.img.xz) and only use a wpa_supplicant.conf from above with the reseau-2.4g info, is it able to connect?

AvidCoder1, 2024-04-16:
@NEOBRETON Is it possible that the SSID you are having troubles getting connected to (SFR_1C78) is a dual 2.4/5ghz network? If so, you have to explicitly tell the Pi Zero W to utilize the 2.4ghz band in order to get connected, otherwise it will attempt to connect to the 5ghz band and fail, as the Pi Zero W (1st gen) does not have 5ghz capability. 

You can achieve this by adding a frequency list to your wpa_supplicant.conf file:

```
ctrl_interface=DIR=/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

freq_list=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 
# Here 

network={
    ssid="YourNetwork"
    psk="VerySecretPassword"
    freq_list=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472
    # OR Here 
}
```

yoidore-banana27, 2025-09-26:
@NEOBRETON 
I encountered the same issue. It seems the hyphen in the SSID causes the problem. I was able to get it working by changing the SSID as follows:

Original SSID: Buffalo-G-XXXX

Modified SSID: BuffaloG

After removing the hyphen, the connection worked without any brcmfmac errors.