motioneye-project/motioneyeos #2061: camera preview stuck on loading animation

SaylorTwift, 2019-08-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427.

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

I am using the following type of camera: mmal
[motioneye (4).log](https://github.com/ccrisan/motioneyeos/files/3525439/motioneye.4.log)

My camera model is: Raspi camera module rev. 1.3.

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3525444/motioneye.4.log)

On a fresh install according to the instructions, with a camera module plugged in, the camera preview is permanently stuck on the loading animation and sometimes upon a reload of the interface, it becomes grey.
According to motioneye.log, it looks like motion isn't launching properly but I don't know how to fix this.

n2mfc, 2019-08-21:
I noticed the same anomaly in Chrome.  This is a recent behavior change.  Works fine in Internet Explorer and as you noted Edge.  Must be a change in Chrome?

SaylorTwift, 2019-08-21:
@n2mfc No, my problem is that the camera doesn't work. Whether I see a loading animation or a "camera not detected" symbol doesn't really matter. I've used this exact pi, this exact camera and chrome before and it worked fine. Then I moved into my new apartment, re-assembled the setup and now my camera isn't detected anymore.

jasaw, 2019-08-22:
There's nothing in your motioneye.log that indicates motion didn't start properly. The only thing suspicious is this line:
> main config file /data/etc/motion.conf does not exist, using default values

If this is a fresh install, that's probably OK. Make sure your camera cable is attached properly. Run `vcgencmd get_camera` to make sure the GPU can talk to the camera.

anupsahaNYC, 2019-08-22:
I'm experiencing the same on a Zero W. I can see the stream on edge, but endless load on Chrome. The only time the stream loads on Chrome is fast network camera mode

SaylorTwift, 2019-08-22:
@jasaw My camera is definitely attached properly. The red camera light is lit up as well. `vcgencmd get_camera` returns `supported=1 detected=1` which I suppose means the camera is supposed to work.
Yet, in the web interface, I can't see an image and instead the camera icon is crossed out on the preview.

SaylorTwift, 2019-08-22:
Does nobody have any ideas? Everything is set up correctly and I've now tried two different cameras as well. Motioneye just shows the crossed out camera symbol despite the camera being detected by the board.

SaylorTwift, 2019-08-22:
Another update: I also tried a different SD card in case the one I was using was corrupted. No luck.

monkeygroover, 2019-08-22:
Same for me, a recent change with chrome on both android and Chrome OS.

SaylorTwift, 2019-08-23:
@monkeygroover  Again, Chrome isn't the problem. The camera just doesn't show a picture, no matter which browser I use.

monkeygroover, 2019-08-23:
Ok, I'll raise a separate issue then, as there also seems to be a problem with video in recent chrome browsers

SaylorTwift, 2019-08-23:
I'm going to close this issue because there obviously is something else wrong here. I will open a new issue with more accurate information.

closed 2019-08-23 by SaylorTwift

vedranius, 2019-08-25:
> I'm going to close this issue because there obviously is something else wrong here. I will open a new issue with more accurate information.

@SaylorTwift  @ccrisan 
I can't find the solution to this problem? Is there any?

I've noticed that it works fine in Firefox, as someone described, but not in Chrome.

Is this problem solved in latest nigthly-dev build? And how to update to it?

Thank you!

ukmercanary, 2019-08-26:
Would  be good to get this working in Chrome I was pulling my hair out trying to figure out if it was just my Mac playing up but no it is just not loading the camera picture in Chrome.  (Possible flash issue? I know Chrome made a change in regards to that recently.)