motioneye-project/motioneyeos #2064: no camera image on all pi devices, all OS versions, all cameras

SaylorTwift, 2019-08-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
20190427 and older

### Board Model

Raspi B Model 3, Zero W

### Camera

I am using the following type of camera: MMAL

My camera model is: 

Pi Camera Module rev. 1.3
Three different camera modules tried

### Network Connection

My motionEyeOS unit is connected to the network via: 
Pi 3: Ethernet
Pi Zero W: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:
None of the logs show any errors. I will provide them if needed.

After opening a different issue that couldn't be resolved, I did a lot of experimentation and still couldn't come up with a solution. Here is the problem:
I have used two Pi Zero Ws and a Pi 3 Model B, two new SD cards, three different camera modules (1.3) and a few different versions of the OS to make sure none of my hardware or software is the issue.
When accessing the web interface of motioneyeos (freshly installed many many times), the camera isn't detected. It shows a crossed out camera symbol. When running `vcgencmd get_camera`, the output is `supported=1 detected=1` which to my knowledge means that the system is detecting the camera.
The web interface is working normally. I've recently moved to a new apartment and I tested the setup at my old place where it worked perfectly. The only thing different is that I have a new router with a new network now.
Could it be possible that my router is somehow suppressing traffic from the video streaming port?

Thank you for helping out. I can't count how much hair I must have pulled out over this so far.

ccrisan, 2019-08-23:
Please try the latest nightly build and see if it fixes your problem.

SaylorTwift, 2019-08-23:
@ccrisan Thanks for helping out. I'd previously tried the latest build and it had the same problem. I'm not kidding though, this issue has just solved itself and I have no idea how or why. I'm now running the latest release without issues.

Thanks again, though. I'll close the issue.

closed 2019-08-23 by SaylorTwift

talormanda, 2019-08-26:
@ccrisan is there a way to get on the nightly build without wiping my OS and starting over with it?

ccrisan, 2019-08-27:
Of course. Just enable prereleases in Expert Settings and then run `fwupdate upgrade yourversion`.

ccrisan, 2019-08-29:
@talormanda of course. We don't currently have an ETA for the next stable release but we're hoping to get it out really soon. There still are some bugs related to recent work that need to be addressed first and I need one or two free weekends for that.

talormanda, 2019-08-29:
Okay thanks for the update