motioneye-project/motioneyeos #2065: Using USB wifi dongle instead of onboard wifi on rpi zero w

mocoloco69, 2019-08-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20180627

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: RPi Camera

My camera model is: V1.3

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Belkin USB WiFi adapter

I am wanting to connect my RPi Zero to a 5G network using a Belkin N150 Wireless USB adapter. I know it works with my Raspbery Pi 3 (running DietPi or OSMC), and on my laptop running Linux Mint 18.1. 

I would like to turn off the onboard wifi and enable the USB wifi. The light doesn't even come on, on the WiFi adapter. I am using a power supply that supplies 5v @ 2000mA. I have used this same power supply on my RPi 3 with no issues, so I don't think it is a power issue. What settings need to be changed for this to work? My Pi does connect to WiFi with the built in onboard wifi, but the connection is spotty as it is far away from the wifi router. I'm trying to connect to a closer router that the 2.4G band no longer works on. The 5G works fine, however. I've tried changing the os.config file (in both partitions that I found it in the /etc directory) to wlan1 instead of wlan0, but that just said "no device found" when it tries to initialize wpa_supplicant.

DuMaM, 2019-08-27:
Motioneyeos is based on [thingos](https://github.com/ccrisan/thingos), because it is a custom made system i suppose that it don't have drivers for your dongle chip.

If you really want to use your dongle please consider installing normal ubuntu or raspbian (there are drivers for your chip) and setup backend and frontend with [motioneye](https://github.com/ccrisan/motioneye/). 

I was using both, options and In my experience motioneyeos is better one (more stable), but It depends on you which solution will you pick.

jasaw, 2019-08-28:
@mocoloco69 Most likely motioneyeos does not have a driver for your wifi dongle. Please attach your dmesg log here. You can also `ifconfig -a` to see if wlan1 exists. If your wifi dongle is not supported, you might be better off running another OS distro as @DuMaM suggested.

mocoloco69, 2019-08-29:
Thank you @DuMaM and @jasaw for your input. When I checked ifconfig -a there was no wlan1. So, I found an old WiFi dongle (the one that came with my Raspberry Pi 3 B - no wifi), and when I plugged that one in, I find wlan1 in ifconfig -a. So, now, how do I set MotionEyeOS to use wlan1 instead of wlan0? Do I just change the os.config file? Or is there more to it?

jasaw, 2019-08-30:
To use `wlan1` instead of `wlan0`, set `OS_WLAN` to `wlan1` in [os.conf](https://github.com/ccrisan/thingos/wiki/os.conf#os_wlan).
Example: Add `OS_WLAN="wlan1"` to /boot/os.conf

mocoloco69, 2019-08-30:
Thank you @jasaw. I had already tried it and gotten it to work, but thank you for verifying that I didn't need to do anything else. I tried the link provided by @DuMaM, but I couldn't find any list of supported WiFi adapters. Just supported boards, like Raspberry Pi, Orange Pi, etc. Is there a list of supported WiFi adapters? I've searched around, but can't find one. I'm not very savvy with GitHub.

DuMaM, 2019-08-31:
@mocoloco69 There is a list with boards my link. Pick yours board and then search inside if driver to your dongle chip is inside drivers list used in board image.

mocoloco69, 2019-09-06:
Ok, so I finally figured out how to read the list of supported drivers. Thank you @DuMaM for that link. I just didn't know how to read it... And, I found the chipset of my WiFi adapter. It is an RTL8188SU, and according to the list of drivers, RTL81XX is supported, and (I believe) enabled by default. So, how to do I get it to recognize that it's there? It recognized the other one right away... Do I need to have it plugged in when I first boot it after a fresh reimage?

DuMaM, 2019-09-07:
@mocoloco69 
There is plenty of ways:
You can log into motionos via ssh and then check [if the driver is loaded.](https://www.cyberciti.biz/faq/linux-show-the-status-of-modules-driver/)
You can read source code config and find out which drivers are loaded. 

How to enable it is showed in link i sent you earlier. 

I earlier show you whole solution. I gave you a rod and tell how to use it. Now you only has to catch the fish. :wink: 
Please, use google and wiki, because how you achieve this is on you side and description of every step is already in Internet. This is part of learning new things :wink:

mocoloco69, 2019-09-08:
Thank you @DuMaM. This is all a bit over my head, and I didn't know where to start. I'm comfortable following instructions, but often times I don't know what to search for, or where. You have given me some good info and good links! Thank you so much for your patience! I will see what I can figure out!