motioneye-project/motioneyeos #2071: ODROID XU4 performance not as good as expected

bobster316, 2019-08-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
motioneyeos-odroidxu4-dev20190820

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
motioneyeos-odroidxu4-dev20190820

### Board Model

I am using the following board/model: Odroid XU4
### Camera

I am using the following type of camera: Network Camera, 

My camera model is: Wyze Cam v2

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, WiFi, 

### Peripherals

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I have recently purchased an Odroid XU4 hoping for better performance than my raspberry pi 4, unfortunately its no better, is there any way to increase the performance of the odroid to get better frame rates

jasaw, 2019-08-29:
Frame rate depends on many things. First of, are you talking about live stream frame rate or frame rate in the recorded videos?

Frame rate is highly affected by your network bandwidth.
- is your XU4 connected via wifi or ethernet?
- wyze cams are all wifi only right? What protocol is used to stream the video? MJPEG or H264 in RTSP? With MJPEG, you won't get high frame rate.
- do wyze cams support 5GHz wifi band? If they do, try switching them over to 5GHz. This should give you more network bandwidth to play with.
- reduce your camera resolution to reduce the amount of data you're pushing into your wifi network.

WRT low frame rate in recorded videos on XU4, I'm working hard to enable hardware accelerated encoder (h264_v4l2m2m) which should increase frame rate significantly. See https://github.com/ccrisan/motioneyeos/issues/1921 if you are interested in my progress or want to help. I don't actually have an XU4 board, so it's a bit hard for me to test.

jasaw, 2019-10-30:
I'm closing this issue because the topic is too general. Any further discussions about the recording frame rate should go to issue: https://github.com/ccrisan/motioneyeos/issues/1921

closed 2019-10-30 by jasaw