motioneye-project/motioneyeos #2072: E-Mail picture attechment does not work

ThomasManthey, 2019-08-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ThomasManthey, 2019-08-28:
motioneyeos-20190413
RaspberryPi Zero WH
On board camera "RPI JT CAN 5MP"
On board network connection via wifi
[motion.log](https://github.com/ccrisan/motioneyeos/files/3552537/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3552538/motioneye.log)
Even if I set the maximum search time there are no pictures at my E-mail.

ccrisan, 2019-08-29:
The message `2019-08-23 18:55:56: [motioneye]    ERROR: timeout waiting for the media listing process to finish` indicates that there are simply too many media files for the email attachment process to find suitable files. Please clean up your media files folder and generally don't keep too many pictures there.

closed 2019-08-29 by ccrisan

ThomasManthey, 2019-09-01:
Witch setting is recommended to attach maybe the last 3 pictures at e-mail?
If I run the camera autonomously is there a other possibility to reduce the number of files than to change the "Preserve Pictures" parameter? Actually it is set to "one month"
By the way this is the first very very stable camera application running on my pizero. Other installations using motion on raspian with different video drivers where crashing every 2 or 3 days. Great work!!